StuartsProjects/SX12XX-LoRa #1: Using with ESP8266

happytm, 2020-02-23:
Thank you for you library.

I was trying to use ESP8266 with EByte E22 900M22S module but I do not know how to connect pins of ESP8266 with this Lora module for simple receiver which after receiving message will publish received data via MQTT. There is no need for TTN network for my application. Would it be same pin connections for same Lora device with ESP8266 for transmitter? 

Please help me.
Thanks.

LoRaTracker, 2020-02-23:
That device appears be a SX1262 with RX and TX enable pins, there is a note in the Readme for the library;

"Some SX126x modules may have RX or TX enable pins, these are currently not supported by the library"

I might add support for those devices in future but I don't have one to test.

happytm, 2020-02-23:
OK.Thank you very much.

ayaghini, 2020-04-01:
I am using the 900M22Ss with:

#define NSS 10                                  //select on LoRa device
#define NRESET 9                                //reset on LoRa device
#define RFBUSY 7                                //SX126X busy pin 
#define DIO1 3

LoRaTracker, 2020-04-01:
I dont know if those pins will work on an ESP8266 and the library only specifically supports Atmega328 style AVRs, as mentioned in the readme. At some point I might publish some examples for ESP32, but its not a priority.

closed 2020-04-01 by LoRaTracker