StuartsProjects/SX12XX-LoRa #2: #include <ProgramLT_Definitions.h>

doidgy, 2020-03-24:
Thank You.

I have searched for days for examples/code that actually work for LoRa without much success, then I find yours and 95% of your code works "out of the box".

The issue I seemed to have was...#include <ProgramLT_Definitions.h> gave me errors....but by removing the LT all was good

Just leaving it here to help others, thank you again.

ayaghini, 2020-03-25:
removing the LT causes: 'LocationPacket' was not declared in this scope.

Many thanks to the author for the well written and documented library though!

LoRaTracker, 2020-03-25:
The versions of the definition files appear to have got swapped when I was adding the tracker application to the SX126X and SX128X examples. 

The correct file is 'ProgramLT_Definitions.h' and has been put back in place in the repository, you can just download that file and place it in the src directory.

doidgy, 2020-03-25:
> removing the LT causes: 'LocationPacket' was not declared in this scope.
> 
> Many thanks to the author for the well written and documented library though!

Not for me, it compiled and has been working

doidgy, 2020-03-25:
> The versions of the definition files appear to have got swapped when I was adding the tracker application to the SX126X and SX128X examples.
>
> The correct file is 'ProgramLT_Definitions.h' and has [...]

Will do thanks

LoRaTracker, 2020-03-25:
The 'ProgramLT_Definitions.h'  is needed as some of the examples use 'Packet Addressing' (see the section in the main Readme). So to ensure that the examples for SX126X and SX127X devices are interoperable, the library was setup to ensure that the different device families share a common set of definitions for packet addressing. 

The tracker receiver example was recently expanded to recognise and display the packets sent from a previous librarys tracker transmitter; 

[](https://github.com/LoRaTracker/Locator2.git)

closed 2020-04-04 by LoRaTracker