StuartsProjects/SX12XX-LoRa #3: PacketError

pekkin, 2020-03-26:
Hello!
I tried to use  this example https://github.com/LoRaTracker/SX12XX-LoRa/tree/master/examples/SX128x_examples/Basics/4_LoRa_Receiver. All checks are good, but I have this mistakes : 

153s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,591,IRQreg,0
153s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,592,IRQreg,0
153s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,593,IRQreg,0
153s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,594,IRQreg,0
154s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,595,IRQreg,0
154s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,596,IRQreg,0
154s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,597,IRQreg,0
154s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,598,IRQreg,0

I find comand which call this mistake:
```
RXPacketL = LT.receive(RXBUFFER, RXBUFFER_SIZE, 60000, WAIT_RX); //wait for a packet to arrive with 60seconds (60000mS) timeout
``` 
Error happens four times per second (not 60 seconds). 
What kind of problem can it be?
I use ESP-WROOM-32.
Thank you!

LoRaTracker, 2020-03-27:
There is no fault or issue with the library. As stated in the main Readme;

### Compatibility

**Fully tested on 3.3V 8Mhz ATMega328P and ATMega1284P only.**
 
It was not the intention to specifically support non-Atmel platforms with the library but several programs have been tested and work on an ESP32 WROOM board and an STM32 Xnucleo board. See the Examples\SX127X_Examples\ESP32 and Examples\SX127X_Examples\STM32
folders.

The Examples\SX127X_Examples\ESP32 folder contains a Readme that explains the changes that were needed to make an example program work on the ESP32. 

At some point I may well add some of the SX1280 examples modified to work on ESP32, STM32 and SAMD21, but its not a prority.

pekkin, 2020-03-27:
> There is no fault or issue with the library. As stated in the main Readme;
>
> ### Compatibility
> **Fully tested on 3.3V 8Mhz ATMega328P and ATMega1284P only.**
>
> It was not the intention to specifically [...]

Hi, thank you for aswer.
We fixed mistakes and got another one:
220s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,182,IRQreg,8000,IRQ_PREAMBLE_DETECTED
I receive all packets from transmitter on reciever. What could be the problem?

LoRaTracker, 2020-03-27:
I cant make sense of what you are reporting and you did not say what mistakes you fixed ?

In any case as I mentioned earlier, I have not yet published any examples for the SX1280 on ESP32, so cannot really comment any changes you have made to the examples programs designed for ATmega328 and 1284.

pekkin, 2020-03-27:
> I cant make sense of what you are reporting and you did not say what mistakes you fixed ?
> 
> In any case as I mentioned earlier, I have not yet published any examples for the SX1280 on ESP32, so cannot really comment any changes you have made to the examples programs designed for ATmega328 and 1284.

Thanks for the quick response!
I forget about DIO1 pin. To be honest I change only SPI pins in this example https://github.com/LoRaTracker/SX12XX-LoRa/tree/master/examples/SX128x_examples/Basics/4_LoRa_Receiver

LoRaTracker, 2020-04-04:
Example transmitter and receiver examples have been added, not the details of the pins used.

pekkin, 2020-04-08:
Hi!
Test your example ESP32  https://github.com/LoRaTracker/SX12XX-LoRa/tree/master/examples/SX128x_examples/ESP32
But the same problem, which i described here 

> Hi, thank you for aswer.
> We fixed mistakes and got another one:
> 220s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,182,IRQreg,8000,IRQ_PREAMBLE_DETECTED
> I receive all packets from [...]

LoRaTracker, 2020-04-08:
Which ESp32 are you using and which pins are you using ?

LoRaTracker, 2020-04-08:
And also please say exactly which LoRa module you are using, its very helpful that I know exactly what your hardware setup is.

closed 2020-09-17 by StuartsProjects