StuartsProjects/SX12XX-LoRa #4: Using multiple SPI devices transmit issue

ayaghini, 2020-04-01:
I said it before but again: Awesome Job and thanks for sharing the library with us!

## [Summary] 
Using SX1262, Arduino Pro Nano if a second device that uses SPI added to the sketch, even after the other released the SPI, this one can not transmit.

## [Explanation] 
- I am using LoraTracker Transmitter and Receiver codes manually combined from SX126XLT. Also, I am using MySensors and NRF24 to connect to my gateway and publish MQTT. 

- I am using PIN 8 for the LoRa CS and 10 for the NRF.

- It seems that MySensors library uses SPI.endTransaction [and I can confirm that it connect to my gateway successfully], yet SX126XLT gets stock in transaction, please find below the DEBUG enabled output:

endWriteSXBuffer()
transmitSXBuffer()
setBufferBaseAddress()
SetPacketParams()
setTxParams()
setDioIrqParams()
setTx()
clearIrqStatus()

- Also, only using the receiver functionality leads to the same problem:

endReadSXBuffer()
L*2,49.31306,-123.03112,60.0m,69,0.69,0,69mS,5320mV,3s,,RSSI,-54dBm,SNR,5dB,Packets,2,Length,31readIrqStatus(),IRQreg,0setMode()
setBufferBaseAddress()
setDioIrqParams()
setRx()
clearIrqStatus()

### [Note1] my sketch works fine without using the MySensors!
### [Note2] the MySensor part simply sends an MQTT message.
### [Note3] Please let me know if posting my code here could help.

Cheers,
AY

LoRaTracker, 2020-04-01:
You can post the code, I dont have your hardware, so cannot run your program directly, but I may spot something. 

That the program hangs, with no timeout, could suggest the lora device is not getting programmed correctly. You can check its settings by placing a printModemSettings() command at a suitable point in the setRX() command in the SX126X.cpp file.

ayaghini, 2020-04-01:
Appreciate the reply. seems no difference with the working one. 

below is the outcome:

printModemSettings()
printDevice()
SX1262,getFreqInt()
915000000hz,SFgetLoRaSF()
12,BWreturnBandwidth()
62500,CR4:getLoRaCodingRate
5,LDRO_getOptimisation
On,SyncWord_0xgetSyncWord
1424getInvertIQ
,IQNormal,Preamble_getPreamble
8

[LoraBasePlusMySensors.zip](https://github.com/LoRaTracker/SX12XX-LoRa/files/4417284/LoraBasePlusMySensors.zip)

LoRaTracker, 2020-04-03:
There is no issue with the library when SPI is being shared with other SPI devices, both an SD card and NRF24 work OK. 

What the issue is with Mysensors, I cannot say, never used that library. 

You need to reduce the problem to the smallest program that shows the problem and check the pins you are using. Maybe Mysensors can help they may have seen issues when other SPI devices are used with their library.

ayaghini, 2020-04-03:
Great that I have confirmation from you for the library working with other SPI devices.

I am under the impression that they use NRF24 library and that library takes care of enabling and disabling SPI every time it uses it, and that's why I was not sure if that the LoRa library do so and opened the issue! 

Thanks for the suggestion, I will put my focus on Mysensors and keep you posted! 

Have a great weekend!
AY

LoRaTracker, 2020-04-03:
> I am under the impression that they use NRF24 library and that library takes care of enabling and disabling SPI every time it uses it, and that's why I was not sure if that the LoRa library do so and opened the issue!

I doubt the NRF24 library disables SPI, that would stop it working. But it should set it up to use the correct parameters its needs every time it uses it. 

The SX12xx library certainly does this as mentioned in the main Readme;

**SPI.begin()**

Standard Arduino library function. Sets up SPI. The library then internally uses;

**SPI.beginTransaction(SPISettings(LTspeedMaximum, LTdataOrder, LTdataMode))**

before every use of the SPI and this function after it;

**SPI.endTransaction()**

ayaghini, 2020-04-04:
## update

After checking both libraries, I found the reset pin of LoRa is shared with NRF default CE pin. 

So basically the library was not able to reset the loRa [IMO]. moving the reset pin of Lora to another pin solved my problem and I can confirm all is working.

Thanks for your help.
AY

LoRaTracker, 2020-04-05:
> I found the reset pin of LoRa is shared with NRF default CE pin.

There was no need to check the library, the pins used are clearly defined in every example and thats on purpose to avoid confusion. 

However when adding another library to a program, such as MySensors, its an elementary to check that the added library does not attempt to use the same pins. 

Do raise issues if there are actual problem found with the SX12XX library, but a problem relating to a third party library is not an issue the SX12XX library.

closed 2020-04-05 by LoRaTracker