StuartsProjects/SX12XX-LoRa #5: SX1280 BLE ble packet

mrninhvn, 2020-04-07:
Hi, Can I send ble packet using sx1280 or not. Thank you veru much.

LoRaTracker, 2020-04-07:
Only LoRa, FLRC and Ranging support and examples have been implemented for SX128X. 

I have no timescale for looking at adding BLE or GFSK.

mrninhvn, 2020-04-07:
> Only LoRa, FLRC and Ranging support and examples have been implemented for SX128X.
> 
> I have no timescale for looking at adding BLE or GFSK.

I see the definition PACKET_TYPE_BLE in SX128XLT_Definitions.h.
Can I using LT.setPacketType(PACKET_TYPE_BLE)?
Do you have any suggestions about modifying libraries to use BLE?

Thank you again.

LoRaTracker, 2020-04-07:
Its not simply a case of changing the packet type. 

You need to read the datasheet, it should have the details needed.

closed 2020-09-17 by StuartsProjects