StuartsProjects/SX12XX-LoRa #6: esp32 PacketError problem on receiver

promistrio, 2020-04-08:
Hello! The same problem like closed issues on ESP32 (basic example for esp32).
I made sure the transmitter was working. Data from HackRF:
[image]
An interesting point was noticed. The message is sent, but it doesn't get broadcast. Check the blue line.

Debug info:
```
20:04:56 Apr  8 2020
V1.1

104_LoRa_Receiver_Detailed_Setup_ESP32 Starting

begin()
SX128XLT constructor instantiated successfully
NSS 5
NRESET 27
RFBUSY 25
DIO1 35
DIO2 -1
DIO3 -1
RX_EN -1
TX_EN -1
RX_EN & TX_EN switching disabled
resetDevice()
checkDevice()
LoRa Device found
setMode()
setRegulatorMode()
setPacketType()
setRfFrequency()
setBufferBaseAddress()
setModulationParams()
SetPacketParams()
setDioIrqParams()

printModemSettings()
printDevice()
SX1280,getFreqInt
2444999936hz,SFgetLoRaSF()
7,BWreturnBandwidth()
406250,CR4:getLoRaCodingRate
5getInvertIQ
,IQNormal,Preamble_getPreamble
12
printOperatingSettings()
printDevice()
SX1280,PacketMode_LORA,Explicit,LNAgain_getLNAgain
LowPowerRX

printRegisters()
Reg    0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
0x900  80 FF 77 41 20 FA BC 13 C1 80 00 00 00 00 00 61 
0x910  9C 44 00 00 00 19 00 00 00 19 87 65 43 21 7F FF 
0x920  FF FF FF 00 70 55 12 50 D0 80 00 C0 5F D2 8F 0A 
0x930  00 C0 00 00 00 24 00 21 28 B0 30 0D 1B 51 63 0C 
0x940  58 0B 32 0A 16 24 6B 96 00 18 00 00 00 00 00 00 
0x950  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x960  00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF 
0x970  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 04 
0x980  00 0B 18 70 00 00 00 4C 00 F0 64 00 00 00 00 00 
0x990  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x9A0  00 08 EC B8 9D 8A E6 66 04 00 00 00 00 00 00 00 
0x9B0  00 08 EC B8 9D 8A E6 66 04 00 00 00 00 00 00 00 
0x9C0  00 16 00 3F E8 01 FF FF FF FF 5E 4D 25 10 55 55 
0x9D0  55 55 55 55 55 55 55 55 55 55 55 55 55 00 00 00 
0x9E0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x9F0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

Receiver ready - RXBUFFER_SIZE 32

receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()4s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,1,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()5s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,2,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()6s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,3,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()7s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,4,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()8s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,5,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()9s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,6,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()10s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,7,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()11s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,8,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()12s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,9,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
setMode()
readIrqStatus()
_RXPacketL: 
0
readPacketRSSI()
readPacketSNR()
readIrqStatus()13s PacketError,RSSI,0dBm,SNR,0dB,Length,readRXPacketL()
0,Packets,0,Errors,10,IRQreg,8000printIrqStatus()
readIrqStatus(),IRQ_PREAMBLE_DETECTED
receive()
setDioIrqParams()
setRx()
clearIrqStatus()
```

I think the error is not related to esp32

promistrio, 2020-04-08:
Sorry, you opened the topic while I was typing this one. I use http://www.ebyte.com/en/product-view-news.aspx?id=286 and pins that you define in settings.h

LoRaTracker, 2020-04-08:
> I think the error is not related to esp32

Can you elaborate, its not clear what you mean ? 

When reporting an issue with the library, its really helpful to me if I know exactly what hardware you are using, in this case which microcontroller and which LoRa module ?

promistrio, 2020-04-08:
this module (sx1280) http://www.ebyte.com/en/product-view-news.aspx?id=286 
dev board based on ESP32-WROOM-32 module. 
[image]

> Can you elaborate, its not clear what you mean ?
The module passed the tests, so the SPI connection works.

promistrio, 2020-04-08:
Another interesting point. The receiver always detects the sent message. But not always on HackRF.

promistrio, 2020-04-08:
Transmitter output
```
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5391
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5392
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5393
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5394
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5395
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5396
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5397
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5398
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5399
10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,25mS  PacketsSent,5400
```

LoRaTracker, 2020-04-08:
SDRs do not always pick up the very short packets, so that the SDR does not see it is not direct evidence that it was not sent.   

With no packet received (or transmitted) a correctly wired ESP32 module will display an RXtimeout message. 

The library and examples work correctly on an ESP32 bare module, but then that is an actual ESP32 WROOM with no additional hardware. Its possible there is a pin conflict with the ESP32 board you are using. These boards are often referred to as ESP32s, buy they can have additional hardware added that causes problems.

promistrio, 2020-04-08:
> With no packet received (or transmitted) a correctly wired ESP32 module will display an RXtimeout message.

Receiver display rxtimeout, but receive and show error every incorrect packet. 

> The library and examples work correctly on an ESP32 bare module, but then that is an actual ESP32 WROOM with no additional hardware. Its possible there is a pin conflict with the ESP32 board you are [...]

May be I will try to connect wires to module without additional hardware.

promistrio, 2020-04-08:
Can you tell me what problems are taking place if there are spi, and GPIO. SPI works in this case. May be I can to fix this.

LoRaTracker, 2020-04-08:
> Receiver display rxtimeout, but receive and show error every incorrect packet.

Can you show me the serial monitor output, that includes both timeout and incorrect packet. Do not reduce or omit from the printout the sequence is important.

promistrio, 2020-04-08:
> > Receiver display rxtimeout, but receive and show error every incorrect packet.
> 
> Can you show me the serial monitor output, that includes both timeout and incorrect packet. Do not reduce or omit from the printout the sequence is important.

```

22:16:37 Apr  8 2020
V1.1

104_LoRa_Receiver_Detailed_Setup_ESP32 Starting

LoRa Device found

SX1280,2444999936hz,SF7,BW406250,CR4:5,IQNormal,Preamble_12
SX1280,PacketMode_LORA,Explicit,LNAgain_LowPowerRX

Reg    0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
0x900  80 FF 77 41 20 FA BC 13 C1 80 00 00 00 00 00 61 
0x910  9C 44 00 00 00 19 00 00 00 19 87 65 43 21 7F FF 
0x920  FF FF FF 00 70 55 12 50 D0 80 00 C0 5F D2 8F 0A 
0x930  00 C0 00 00 00 24 00 21 28 B0 30 0D 1B 51 63 0C 
0x940  58 0B 32 0A 16 24 6B 96 00 18 00 00 00 00 00 00 
0x950  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x960  00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF 
0x970  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 04 
0x980  00 0B 18 70 00 00 00 4C 00 F0 64 00 00 00 00 00 
0x990  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x9A0  00 08 EC B8 9D 8A E6 66 04 00 00 00 00 00 00 00 
0x9B0  00 08 EC B8 9D 8A E6 66 04 00 00 00 00 00 00 00 
0x9C0  00 16 00 3F E8 01 FF FF FF FF 5E 4D 25 10 55 55 
0x9D0  55 55 55 55 55 55 55 55 55 55 55 55 55 00 00 00 
0x9E0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x9F0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

Receiver ready - RXBUFFER_SIZE 32

3s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,1,IRQreg,8000,IRQ_PREAMBLE_DETECTED
4s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,2,IRQreg,8000,IRQ_PREAMBLE_DETECTED
5s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,3,IRQreg,8000,IRQ_PREAMBLE_DETECTED
6s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,4,IRQreg,8000,IRQ_PREAMBLE_DETECTED
7s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,5,IRQreg,8000,IRQ_PREAMBLE_DETECTED
8s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,6,IRQreg,8000,IRQ_PREAMBLE_DETECTED
9s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,7,IRQreg,8000,IRQ_PREAMBLE_DETECTED
10s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,8,IRQreg,8000,IRQ_PREAMBLE_DETECTED
11s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,9,IRQreg,8000,IRQ_PREAMBLE_DETECTED
13s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,10,IRQreg,8000,IRQ_PREAMBLE_DETECTED
14s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,11,IRQreg,8000,IRQ_PREAMBLE_DETECTED
15s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,12,IRQreg,8000,IRQ_PREAMBLE_DETECTED
16s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,13,IRQreg,8000,IRQ_PREAMBLE_DETECTED
17s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,14,IRQreg,8000,IRQ_PREAMBLE_DETECTED
18s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,15,IRQreg,8000,IRQ_PREAMBLE_DETECTED
19s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,16,IRQreg,8000,IRQ_PREAMBLE_DETECTED
20s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,17,IRQreg,8000,IRQ_PREAMBLE_DETECTED
21s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,18,IRQreg,8000,IRQ_PREAMBLE_DETECTED
22s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,19,IRQreg,8000,IRQ_PREAMBLE_DETECTED
23s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,20,IRQreg,8000,IRQ_PREAMBLE_DETECTED
24s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,21,IRQreg,8000,IRQ_PREAMBLE_DETECTED
25s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,22,IRQreg,8000,IRQ_PREAMBLE_DETECTED
26s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,23,IRQreg,8000,IRQ_PREAMBLE_DETECTED
27s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,24,IRQreg,8000,IRQ_PREAMBLE_DETECTED
28s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,25,IRQreg,8000,IRQ_PREAMBLE_DETECTED
29s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,26,IRQreg,8000,IRQ_PREAMBLE_DETECTED
30s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,27,IRQreg,8000,IRQ_PREAMBLE_DETECTED
31s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,28,IRQreg,8000,IRQ_PREAMBLE_DETECTED
32s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,29,IRQreg,8000,IRQ_PREAMBLE_DETECTED
33s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,30,IRQreg,8000,IRQ_PREAMBLE_DETECTED
34s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,31,IRQreg,8000,IRQ_PREAMBLE_DETECTED
35s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,32,IRQreg,8000,IRQ_PREAMBLE_DETECTED
36s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,33,IRQreg,8000,IRQ_PREAMBLE_DETECTED
37s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,34,IRQreg,8000,IRQ_PREAMBLE_DETECTED
[... 18 lines cut ...]
57s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,53,IRQreg,8000,IRQ_PREAMBLE_DETECTED
58s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,54,IRQreg,8000,IRQ_PREAMBLE_DETECTED
59s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,55,IRQreg,8000,IRQ_PREAMBLE_DETECTED
60s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,56,IRQreg,8000,IRQ_PREAMBLE_DETECTED
61s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,57,IRQreg,8000,IRQ_PREAMBLE_DETECTED
62s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,58,IRQreg,8000,IRQ_PREAMBLE_DETECTED
63s RXTimeout
63s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,59,IRQreg,8000,IRQ_PREAMBLE_DETECTED
[... cut ...]
74s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,69,IRQreg,8000,IRQ_PREAMBLE_DETECTED
75s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,70,IRQreg,8000,IRQ_PREAMBLE_DETECTED
76s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,71,IRQreg,8000,IRQ_PREAMBLE_DETECTED
77s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,72,IRQreg,8000,IRQ_PREAMBLE_DETECTED
78s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,73,IRQreg,8000,IRQ_PREAMBLE_DETECTED
79s PacketError,RSSI,0dBm,SNR,0dB,Length,0,Packets,0,Errors,74,IRQreg,8000,IRQ_PREAMBLE_DETECTED

```

promistrio, 2020-04-08:
1. How to check whether the payload changes in the message on the transmitter? I swapped two modules and the radio transmission still works. May be the problem is not with SPI.
2. Tomorrow I can test on a bluepill stm32f103 and with arduino library
3. I still question the poor quality of the esp32 devboard. SPI transmission works during the test, works for transmission, but does not work during radio reception. GPIO pins function without complaints.

LoRaTracker, 2020-04-09:
The examples work reliably on a bare bones ESP32, so I suspect your issue is that your ESP32 board has some additional haedware that is interfering with the pins being used. 

Cannot comment on whether the examples will work on an Bluepill, not tested.

promistrio, 2020-04-09:
Could you send your circuit for esp32?

LoRaTracker, 2020-04-09:
[https://github.com/LoRaTracker/SX12XX-LoRa/blob/master/pictures/ESP32%20Bare%20Bones%20circuit.pdf](url)

promistrio, 2020-04-15:
I'm sure the transmitter is sending data, because the height on the SDR depends on the size of the message. For example, there are two simultaneous transmitters with different message lengths.
https://youtu.be/YRTEGxApDOM

promistrio, 2020-04-25:
BluePill works without problems! It's really ESP32 problem. I suggest adding an entry about this to the ReadMe.

LoRaTracker, 2020-04-26:
There already is a note about potential problems ESP32s in the readme. 

There is no generic problem with the code not working on ESP32, bu there may well be an issue with your particular ESP32 or how it is wired. 

There are too many ESP32 board varients for it to be practical to test them all, there are over 60 supported by the Arduino IDE.

closed 2020-04-26 by LoRaTracker