StuartsProjects/SX12XX-LoRa #7: ESP32 Deep Sleep code?

qubi-dev, 2020-04-09:
I have a DRF1262T module and tested out different libraries RadioLib and beegee-tokyo. The current is too high using an ESP32 + SX1262 DRF1262T module. ESP32 alone , current is fine in sleep mode. Do you have any examples with ESP32 that put the RF module into deep sleep mode?

Thanks!

LoRaTracker, 2020-04-09:
There is an example in the SX127X folder for using ESP32 light sleep mode. The LT.setSleep(); command used there puts the LoRa device into deep sleep. Same command works for SX126X.

qubi-dev, 2020-04-10:
I used your library, used the same pinouts you have in your program and my output is still in sleep mode 1.
Here is the code:

/*******************************************************************************************************
  lora Programs for Arduino - Copyright of the author Stuart Robinson - 04/04/20

  This program is supplied as is, it is up to the user of the program to decide if the program is
  suitable for the intended purpose and free from errors.
*******************************************************************************************************/

/*******************************************************************************************************
  Program Operation - This is a program that demonstrates the detailed setup of a LoRa test receiver.
  The program listens for incoming packets using the LoRa settings in the 'Settings.h' file. The pins
  to access the lora device need to be defined in the 'Settings.h' file also.

  There is a printout on the Arduino IDE Serial Monitor of the valid packets received, the packet is
  assumed to be in ASCII printable text, if it's not ASCII text characters from 0x20 to 0x7F, expect
  weird things to happen on the Serial Monitor. The LED will flash for each packet received and the
  buzzer will sound, if fitted.

  Sample serial monitor output;

  7s  Hello World 1234567890*,CRC,DAAB,RSSI,-52dBm,SNR,9dB,Length,23,Packets,5,Errors,0,IRQreg,50

  If there is a packet error it might look like this, which is showing a CRC error,

  968s PacketError,RSSI,-87dBm,SNR,-11dB,Length,23,Packets,613,Errors,2,IRQreg,70,IRQ_HEADER_VALID,IRQ_CRC_ERROR,IRQ_RX_DONE

  Serial monitor baud rate is set at 9600.
*******************************************************************************************************/

#define Program_Version "V1.1"

#include <SPI.h>                                 //the lora device is SPI based so load the SPI library
#include <SX126XLT.h>                            //include the appropriate library   
#include "Settings.h"                            //include the setiings file, frequencies, LoRa settings etc   

SX126XLT LT;                                     //create a library class instance called LT

uint32_t RXpacketCount;
uint32_t errors;

uint8_t RXBUFFER[RXBUFFER_SIZE];                 //create the buffer that received packets are copied into

uint8_t RXPacketL;                               //stores length of packet received
int8_t  PacketRSSI;                              //stores RSSI of received packet
int8_t  PacketSNR;                               //stores signal to noise ratio (SNR) of received packet

#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  5        /* Time ESP32 will go to sleep (in seconds) */

void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

void loop()
{

[... 146 lines cut ...]
  First we configure the wake up source
  We set our ESP32 to wake up every 5 seconds
  */
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
 // Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) +
 // " Seconds");

  // Serial.println("Going to sleep now");
  Serial.flush(); 
  esp_deep_sleep_start();
}

void listenMessages(){
    RXPacketL = LT.receive(RXBUFFER, RXBUFFER_SIZE, 60000, WAIT_RX); //wait for a packet to arrive with 60seconds (60000mS) timeout

  PacketRSSI = LT.readPacketRSSI();              //read the recived RSSI value
  PacketSNR = LT.readPacketSNR();                //read the received SNR value

  if (RXPacketL == 0)                            //if the LT.receive() function detects an error, RXpacketL is 0  
    packet_is_Error();  
  else  
    packet_is_OK();
  
  Serial.println();
}

LoRaTracker, 2020-04-10:
The code you have written is attempting to put the ESP32 into deep sleep, and this as I recall, due to issues with the ESP32, causes the LoRa device to wakeup. There is no fault or issue with the library or examples but there is a problem with the ESP32.  

I may at some point see if there is a work around for the technical problems with ESP32 deep sleep. but it was not my intention to provide technical support for the ESP32 so this is not a priority.

qubi-dev, 2020-04-10:
I understand, I found other threads and currently trying to troubleshoot. What I found out is this, The default SPI pins are not part of the RTC pins and when the ESP32 goes to deep sleep, loses the pin state. I am testing remapping right now and setting pin states manually before sleep mode and see if the current is any better.

LoRaTracker, 2020-04-10:
Most all modules that are referred to as ESP32s are boards with an ESP32 module plus a bunch of additional components that make low current deep sleep impossible. 

The only real way to guarantee a low deep sleep current is to use an actual ESP32 Wroom module in bare bones mode and not use one of the ESP32 Wroom + lots of other components boards.

There is an explanation of some of the problems with deep sleep on ESP32 boards here, but it only applies to bare bones modules;

https://www.loratracker.uk/esp32-shield-adding-a-lora-device/

qubi-dev, 2020-04-10:
I am using a bare bones WROOM 32 board on a PCB I made. I can get <10uA in sleep mode perfectly fine :)

qubi-dev, 2020-04-10:
I disconnected all the wires except for VCC/GND and I am still getting the higher current draw from the RF Module for some reason. It might not be the SPI pins like I originally thought.

qubi-dev, 2020-04-11:
I tried E-Byte and DORJI. Same issue. So, I took away the sleep code for ESP32 and kept only the sleep code for LoRA module and I am getting either 450nA or 2.5uA. (Need a better scope) Either way I see the issue is only when I put the ESP32 itself to sleep and has to do with pins in different states. I will troubleshoot further to figure out the solution. I guess this can close. Thanks

closed 2020-04-11 by qubi-dev

johnnytolengo, 2021-09-27:
hi @qubi-dev  what pin are you using to fully control the DRF1262T apart from the SPI pins?  
How to do initialize the library with tht DRF1262T?

I am no sure about the DIO2 pin that controls the TX/RX.