StuartsProjects/SX12XX-LoRa #9: sendLocationBinary sending Location package

ayaghini, 2020-04-22:
Dear Stuart,

Not sure if it fits into "issues" but I noticed that the sendLocationBinary in example 23 provided in the library for 126x sends all parameters [i.e. PacketType == LocationPacket]. 

That does not agree with the PacketType == LocationBinaryPacket definition on the receiver side.

All the best,
AY

LoRaTracker, 2020-04-22:
Not sure what your saying here ?

PacketType == LocationPacket and PacketType == LocationBinaryPacket are different packet types with different contents.   

Does the example work correctly ?

ayaghini, 2020-04-22:
While I believe the "Binary packet"  is designed to be a short package only contains location data, the "sendLocationBinary" function in the code sends below parameters:

  LT.writeUint8(LocationPacket);              //indentify type of packet
  LT.writeUint8(Broadcast);                   //who is the packet sent too
  LT.writeUint8(ThisNode);                    //tells receiver where is packet from
  LT.writeFloat(Lat);                         //add latitude
  LT.writeFloat(Lon);                         //add longitude
  LT.writeFloat(Alt);                         //add altitude
  LT.writeUint8(TXSats);                      //add number of satellites
  LT.writeUint32(Hdop);                       //add hdop
  LT.writeUint8(TXStatus);                    //add tracker status
  LT.writeUint32(fixtime);                    //add GPS fix time in mS
  LT.writeUint16(TXVolts);                    //add tracker supply volts
  LT.writeUint32(millis());                   //add uptime in mS

It's my understanding that those belong to the normal location packet. 

As I said nothing crazy, still in long range test might help to have them corrected.

Thanks Stuart,
AY

LoRaTracker, 2020-04-22:
So in essense your comment is that the function name, 'sendLocationBinary' in one example program should be changed ?

I just dont see how the actual name of the function call in any way affects the operation of the program ?

closed 2020-09-17 by StuartsProjects