StuartsProjects/SX12XX-LoRa #11: Required pins for sx1280

yo90bosses, 2020-04-24:
I have a sx1280 from ebyte and im wondering if i actually need to use the TX_EN and RX_EN pins as other moduls dont have them. Also in a circuit diagram the DIO pins 1 to 3 arent connected to the MCU. Can i omit those too? Im really low on pins for my project and need to save some for others things.

LoRaTracker, 2020-04-24:
The main readme says on this subject;

"Some of the SX128x modules do have RX or TX enable pins, such as the Ebyte modules, these are supported by the library, and you need to define the pins RX_EN and TX_EN pins used, otherwise leave unused by defining them as -1"

So yes you need to use them.

LoRaTracker, 2020-04-24:
For a definative answer, since you know exactly which 'ebyte' module you have, refer to the devices datasheet.

LoRaTracker, 2020-04-24:
> Also in a circuit diagram the DIO pins 1 to 3 arent connected to the MCU.

The library does require the use of DIO1.

DI02 and DIO3 are marked in all SX128X examples as 'not used' so you dont need to connect them.

closed 2020-09-17 by StuartsProjects