StuartsProjects/SX12XX-LoRa #13: Investigate adding internal SX1278 temperature sensor........

Cloolalang, 2020-06-08:
Hey Im also taking a look at temperature compensation in narrow bandwidth modes.
Im seeing about 300Hz/Celsius drift with my RA-02.
Im thinking for using two loops to track the temperature drift issue, outer loop to measure on-board temp and then adjust XTAL Freq accordingly.
Then use FEI error measurement to do the final fine-tune when my radio is connected to another (a master/slave frequency control)
I didnt much like the internal temperature sensor so prefer to go and try an external sensor.
Cheers
P
ZS1dBm

LoRaTracker, 2020-06-09:
Seems a lot of work when the SX126X modules have the problem sorted, they include a TCXO. 

You can operate them all the way down to 7800hx bandwidth, even with significant temperature variations between TX and RX.

LoRaTracker, 2020-06-11:
I have tested NiceRF and Dorji modules. Both very temperature stable.

Cloolalang, 2020-06-11:
Thanks for the information,. Ill give it a go with a Dorji module.
Going to have a go at the LoRa-APRS here in Cape Town.
Cheers
P

closed 2020-07-12 by Cloolalang

Cloolalang, 2020-07-12:
I have my new sx1262 modules, so I see your library is very well documented and so Im keen to start work on it. My question is, as Im coming from using the RadioHead/mysensors, I would like to know if I can pass uart/serial over your SX12XX?
That way I keep the transport protocol for the sensors etc but get a much more up-to-date radio link.
I like the Mysensors protocol as it interfaces pretty well with MQTT and the IOT controllers Im using (Home assistant).
73 Patrick
ZS1dbm

LoRaTracker, 2020-07-12:
> I would like to know if I can pass uart/serial over your SX12XX?
 
Not sure what your asking here.

Cloolalang, 2020-07-12:
Hey, oh sorry, I wasnt very clear.
Currently I use RFM95 with the Mysensors library, I would like to get on with using the SX1262 radios, but Mysensors dont yet have support for the sx1262 (the library based on the RadioHead Library, so its getting a bit old.
I would like to know If I can use the SX12xx library to send and receive the Mysesnors serial protocol.
https://www.mysensors.org/download/serial_api_20
As I have loads of sensors and a controller setup using that transport and tree network with some repeaters.
I guess I could just use on of the Ebyte Uart Lora modules, however I would like to try the <20K Lora BW Phy to be able to ramp up to higher on-channel duty cycles and improve the range and I cant see how to fig behind the UART  on those modules and fiddle with the Phy settings.
I have my Mysensors working pretty good at 62KHz/SF12, even with the long on-air times, Its OK as I can run a dual-speed network, Fast for near nodes, and long/slow for far and point-point links.
Im at a point where I either try to port in some SX1262 to mysensors or just try sticking wioth SX127X and drop in some temperature compensation to narrow up the BW.
Cheers
P

LoRaTracker, 2020-07-13:
I dont see whay you could not use the SX12xx library as a basis to build a serial interface to Mysensors.  Fair amount of programming effort I guess.

Cloolalang, 2020-07-16:
Cool, Thanks.
In the mean-time I am taking a look at basic temperature compensation for the SX1278, using the RA-02.
I'm seeing that the Xtal on-board is 1.4KHz off tune as is at 24'C with the module I have here.
After correcting for that, between 10C and 60C its about 100Hz/C drift, does that figure sound likely to you?  If so, Im going monitor the chip temp with a DS18B20, smooth the values and probably step the frequency by 1KHz per 10C change..(If Im correct, at 7.8KHz BW we cant be off more than 1.95KHz so stepping half that should be fine enough, what do you think?
All the RA-02 modules in the system will need initial calibration but that ok for me. (im using an RTL-SDR and a local Digital TV carrier as a reference tone at the moment, GPSD oscillator would be better).
I will go with this until someone smarter than me write the SX1262 library for Mysensors.
P

LoRaTracker, 2020-07-17:
I have heard of people trying temperature compensation, never tried it myself, so cannot comment further. Neither do I use MySensors, so for me I would just use a SX126X if frequency stability was important.