StuartsProjects/SX12XX-LoRa #14: Error in SX126XLT.cpp

MC123T, 2020-06-15:
Hey, i think i found a bug in SX126XLT.cpp

my Setup : Semtech SX1261MB2BAS Shield  +  Arduino Pro mini 3.3V 8Mhz
my Problem: the program stayed in DI01 checking  loop forever.
`while (!digitalRead(_TXDonePin));                                //Wait for DIO1 to go high` (line 1362)

my Solution: 
`  while (!digitalRead(_TXDonePin)){    //Wait for DIO1 to go high
	  if (readIrqStatus() & IRQ_RX_TX_TIMEOUT )                        //check for timeout
	  {
		return 0;
	  }
	  else
	  {
		return _TXPacketL;
	  }
  }  `

(itsjust the bracket position)

i hope my report helps!

LoRaTracker, 2020-06-15:
Does not help really. 

If you check a few lines earlier in SX126X.cpp, 1354, you will see that _TXDonePin (DIO1) is already set to go active on a timeout;

`setDioIrqParams(IRQ_RADIO_ALL, (IRQ_TX_DONE + IRQ_RX_TX_TIMEOUT), 0, 0);   //set for IRQ on TX done and timeout on DIO1`

Are you saying that the SX126XLT::transmit() function is being called with a non-zero value for timeout (3rd parameter) yet no timeout occurs ?

MC123T, 2020-06-15:
okay, lets see what the problem is. 
I thougt, the timeout interrupts should be checked while the TXdone pin is still Low.
That was the reason why i changed the bracket position.

I am using the example 103 LoRa Transmitter with detailed Setup.
the transmit is beeing called like this : 
`LT.transmit(buff, TXPacketL, 1000, TXpower, WAIT_TX`

(WAIT_TX = 1)

If iam using the unmodified libary file (SX126XLT.cpp) my program stays in the mentioned loop forever. readIrqStatus() returns 0 and IRQ_RX_TX_TIMEOUT is 512.  : 

` while (!digitalRead(_TXDonePin)){  //Wait for DIO1 to go high
	  Serial.println(readIrqStatus());
	  Serial.println(IRQ_RX_TX_TIMEOUT);
  }	 

  if (readIrqStatus() & IRQ_RX_TX_TIMEOUT )                        //check for timeout
  {
    return 0;
  }
  else
  {
    return _TXPacketL;
  }`

If i change the code in the transmit function to this (for debugging) : 
`  while (!digitalRead(_TXDonePin)){    //Wait for DIO1 to go high
	  Serial.println("not_TXDonePin");
	  Serial.println(readIrqStatus());
	  Serial.println(IRQ_RX_TX_TIMEOUT);
	  if (readIrqStatus() & IRQ_RX_TX_TIMEOUT )                        //check for timeout
	  {
		return 0;
	  }
	  else
	  {
		return _TXPacketL;
	  }
  } `

my output looks like this : 
`Transmitter ready
2dBm Packet> Hello World 1234567890*not_TXDonePin
0        [= readIrqStatus() ]
512    [ = IRQ_RX_TX_TIMEOUT ]
  BytesSent,23  CRC,DAAB  TransmitTime,2mS  PacketsSent,1
`

looks like the program is going through the loop 1 time ..

i hope this is more useful:)

LoRaTracker, 2020-06-15:
Looks like your hardware setup is not reading the DIO1 pin correctly.
And because you have altered the code, the printout that the packet has been sent is not valid, you dont know if a packet was sent or not.

LoRaTracker, 2020-06-15:
Its also possible your SX1276MB1MAS boards need some particular setup to make it work. 

The library was not tested against that particular board. Only the NiceRF and Dorji modules were tested, as mentioned in the readme.

MC123T, 2020-06-16:
hmm. I dont i think i understood your code correctly .. how should the program be able to get beyond line 1362 if its stuck in that loop? It would never check the Irq Status and for the timeout?
I thought it should check the statement in line 1364 while its waiting for TXDonePin to go high?

LoRaTracker, 2020-06-16:
Note that the code a few lines earlier (1354) sets DIO1 to go active on TXdone or TXtimeout. 

There is then no need to then keep reading the IRQstatus to check for a timeout, the SX126X will set DIO1 high when there is a timout.

LoRaTracker, 2020-06-16:
I suspect the issue you are seeing with these boards is to do with the TCXO setup, you would need to check the datasheet of the SX1261MB2BAS for details. 

You might need to create your own special and additional code, to make those modules work. Or just use the code that Semtech supply.

MC123T, 2020-06-16:
> I suspect the issue you are seeing with these boards is to do with the TCXO setup, you would need to check the datasheet of the SX1261MB2BAS for details.

That was a good hint. the SX1261MB2BAS Shield is using a XTAL oscilator and not a TCXO.
A reference design explanation from semtech can be found here : https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R000000HSQj/MmehlfBHMs5N326wBVRKw68_lEoDBCVcB71X7BUQXYY

now i commented LT.setDIO3AsTCXOCtrl(TCXO_CTRL_3_3V); out of the setup routine and i looks like its working! at least the transmit function isnt stuck in the loop anymore.. I dont know if its actually transmitting or not.. (i will setup a receiver soon...)

is there any other calibration of the TCXO happening that i should not do ? 

the datasheet of SX1261 says : 

> In case of POR or when the device is recovering from Sleep mode in cold start mode, the image calibration is performed as
part of the initial calibration process and for optimal image rejection in the band 902 - 928 MHz. However at this stage the
internal state machine has no information whether an XTAL or a TCXO is fitted. When the 32 MHz clock is coming from a
TCXO, the calibration will fail and the user should request a complete calibration after calling the function
SetDIO3AsTcxoCtrl(...).

Datasheet  (page 55, section 9.2) : https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R000000Q2Kq/31nDblsvXMY94atnuZBurFOXI4KAa2uKsr26Vyh0snc  

You mentioned that only NiceRF and Dorji modules have been tested. Do both of them have a TCXO?

LoRaTracker, 2020-06-16:
Yes the NiceRF and Dorji modules have TXCOs. 

There are tests you need to carry out, to be sure that the 'new' code to support the SX1261MB2BAS is correctly configured. It may not be straight forward to deal with any issues that arise, and trying to resolve these remotly may occupy a great deal of (my) time. There seems to be little point in changing the SX126X Arduino Library to support the  SX1261MB2BAS, this device is from 3 - 6 times the price of alternative SX126X devices and is unlikley to appeal to many Arduino users.

closed 2020-06-16 by LoRaTracker