StuartsProjects/SX12XX-LoRa #15: NOTE Generic Solution regarding RF Solution's LAMBDA range not working!

the-drums, 2020-06-18:
After a lot of testing/debugging, finally got the RF Solutions LAMBDA62C modules to work!

The previous remark regarding using LDO instead of DC-DC appears to **not** be the issue., DC-DC circuit is fine and works properly.

**The solution:**
DONT USE command: "SetDIO3AsTCXOCtrl"
DONT USE command: "SetDIO2AsRfSwitchCtrl" when you have a REV2 Module!

As it appears from a diagram I've retrieved somewere:
Using command: "datasheet: --> 13.3.6 "SetDIO3AsTCXOCtrl" to control TXCO reference voltage causes the malfunction: BUSY_LINE stays high @ the upfollowing command datasheet --> 13.1.12 "Calibrate Function", indicating something is wrong.
This simply is caused by DIO3 not been connected internally to anything!.
After disabling this command,  Calibrate function works  && RX / TX works on all modules!

Sidenote: 
RF Solution has 2 Revisions (one is not yet available):
REV2 (Datasheet: LAMBDA62-2) shows no diagram Important! --> DIO2 is NOT connected to RF_switch
REV3 (Datasheet: LAMBDA62-3) shows a diagram with DIO2 connected to RF_switch.

This concludes that:
REV2 --> Command: datasheet --> 13.3.5 "SetDIO2AsRfSwitchCtrl"  **should NOT be used!**
REV3 -->Command: datasheet --> 13.3.5 "SetDIO2AsRfSwitchCtrl"  can be used!

Hopefully with this I save some people a lot of time figuring it out.

Kind regards, 
Paul

LoRaTracker, 2020-06-18:
Thanks for the comments. 

It is unforunate that the SX126X has introduced a new level of complexity to the base configuration of the module, so many variations. In part the range of the possibilities was the reason for publishing examples 103 and 104 which show the full detail of whats included in the  setupLoRa() function. 

I have been thinking of adding a readme to cover some of the issues discovered with different modules. However before I would add a module to the document I would need to carry out my own checks on the module. You will understrand that I am not going to be buying a few copies of each new SX126X module that is produced. So if you want a module added to a support\tested list, donations of modules will be welcomed.

the-drums, 2020-06-18:
Thank you for your reply.
Also posted this note in:
https://github.com/jgromes/RadioLib/issues/153#issuecomment-645914608
Some additional comments were made.
Also to do use LDO and not DC-DC.

I understand you can't test every module this would be a bit overkill, but says a lot you would even consider it! Though it would surely help out.
But for now hopefully some other users will read this that have purchased the modules  and can skip the troubleshooting for now :).

closed 2020-06-18 by the-drums