StuartsProjects/SX12XX-LoRa #16: DIO1 always low

chegewara, 2020-07-17:
Hi,
i am trying to run esp32 S2 examples
https://github.com/StuartsProjects/SX12XX-LoRa/tree/master/examples/SX128x_examples/ESP32/Basics

but no matter what timeout i will set DIO1 is always low (both, transceiver and receiver):
`RXPacketL = LT.receive(RXBUFFER, RXBUFFER_SIZE, 6, 1);`

Chip is NiceRF SX1280 and i am trying to run LoRa.

Here is log that shows SPI and LoRa should works:
```
21:16:19.955 -> 21:16:14 Jul 17 2020
21:16:19.955 -> V1.1
21:16:19.955 -> 
21:16:19.955 -> 104_LoRa_Receiver_Detailed_Setup_ESP32 Starting
21:16:19.955 -> 
21:16:20.054 -> LoRa Device found
21:16:21.544 -> 
21:16:21.544 -> SX1280,2444999936hz,SF7,BW406250,CR4:5,IQNormal,Preamble_12
21:16:21.544 -> SX1280,PacketMode_LORA,Explicit,LNAgain_LowPowerRX
21:16:21.544 -> 
21:16:21.544 -> Reg    0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
21:16:21.577 -> 0x900  80 FF 77 41 20 FA BC 13 C1 80 00 00 00 00 00 61 
21:16:21.577 -> 0x910  9C 44 00 00 00 19 00 00 00 19 87 65 43 21 7F FF 
21:16:21.577 -> 0x920  FF FF FF 00 70 55 12 50 D0 80 00 C0 5F D2 8F 0A 
21:16:21.577 -> 0x930  00 C0 00 00 00 24 00 21 28 B0 30 0D 1B 51 63 0C 
21:16:21.577 -> 0x940  58 0B 32 0A 16 24 6B 96 00 18 00 00 00 00 00 00 
21:16:21.577 -> 0x950  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
21:16:21.610 -> 0x960  00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF 
21:16:21.610 -> 0x970  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 04 
21:16:21.610 -> 0x980  00 0B 18 70 00 00 00 4C 00 F0 64 00 00 00 00 00 
21:16:21.610 -> 0x990  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
21:16:21.610 -> 0x9A0  00 08 EC B8 9D 8A E6 66 04 00 00 00 00 00 00 00 
21:16:21.610 -> 0x9B0  00 08 EC B8 9D 8A E6 66 04 00 00 00 00 00 00 00 
21:16:21.610 -> 0x9C0  00 16 00 3F E8 01 FF FF FF FF 5E 4D 25 10 55 55 
21:16:21.643 -> 0x9D0  55 55 55 55 55 55 55 55 55 55 55 55 55 00 00 00 
21:16:21.643 -> 0x9E0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
21:16:21.643 -> 0x9F0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
21:16:21.643 -> 
21:16:21.643 -> 
21:16:21.643 -> Receiver ready - RXBUFFER_SIZE 32
21:16:21.643 -> 
```

Register test result:
```
21:26:13.183 -> Reset device
21:26:13.216 -> Registers at reset
21:26:13.216 -> Reg    0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
21:26:13.216 -> 0x900  80 0C 7B 02 20 FA C0 00 00 80 00 00 00 00 00 FF 
21:26:13.249 -> 0x910  FF FF 00 00 00 19 00 00 00 19 87 65 43 21 7F FF 
21:26:13.249 -> 0x920  FF FF FF 0C 70 37 0A 50 D0 80 00 C0 5F D2 8F 0A 
21:26:13.249 -> 0x930  00 C0 00 00 00 24 00 21 28 B0 30 09 1A 59 70 08 
21:26:13.282 -> 0x940  58 0B 32 0A 14 24 6A 96 00 18 00 00 00 00 00 00 
21:26:13.282 -> 0x950  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
21:26:13.315 -> 0x960  00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF 
21:26:13.315 -> 0x970  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 04 
21:26:13.348 -> 0x980  00 0B 18 70 00 00 00 4C 00 F0 64 00 00 00 00 00 
21:26:13.381 -> 0x990  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
21:26:13.381 -> 0x9A0  00 08 EC B8 9D 8A E6 66 04 00 00 00 00 00 00 00 
21:26:13.381 -> 0x9B0  00 08 EC B8 9D 8A E6 66 04 00 00 00 00 00 00 00 
21:26:13.415 -> 0x9C0  00 16 00 3F E8 01 FF FF FF FF 5E 4D 25 10 55 55 
21:26:13.415 -> 0x9D0  55 55 55 55 55 55 55 55 55 55 55 55 55 00 00 00 
21:26:13.448 -> 0x9E0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
21:26:13.448 -> 0x9F0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
21:26:13.481 -> 
21:26:13.481 -> 
21:26:13.481 ->  Frequency at reset 2495996672hz
21:26:13.481 -> Change Frequency to 2445000000hz
21:26:13.481 ->       Frequency now 2444999936hz
21:26:13.481 -> 
21:26:13.481 -> Reg    0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
21:26:13.481 -> 0x900  80 0C 7B 02 20 FA BC 13 C1 80 00 00 00 00 00 61 
```

Thanks

StuartsProjects, 2020-07-18:
The example given and the pinouts given do work correctly on an ESP32. 

None of the examples, not the library have been checked against the ESP32-S2, maybe there is some difference in the new S2 device. .

chegewara, 2020-07-18:
As you can see there is no problem with S2. I am new to LoRa, but log outputs looking good i think.
What i am having problem with is DIO1 not rising IRQ even on timeout and i have hooked up 2 modules with the same behavior.
So i thought i may have missing something in settings.

Today or tomorrow i will try to connect sx1262 modules and let you know.

Thanks for support

IoTThinks, 2020-07-18:
What event you want to catch with DIO1?

chegewara, 2020-07-18:
`setDioIrqParams(IRQ_RADIO_ALL, (IRQ_TX_DONE + IRQ_RX_TX_TIMEOUT), 0, 0);` 
This is part of library code when transmit or receiving, which means it TX/RX done + TIMEOUT. I even changed it to all events, but DIO1 is dead.

Thanks

StuartsProjects, 2020-07-18:
Well since the supplied example code does work on an ESP32 with the NiceRF 1280 module mentioned, this suggests something different in your setup. 

Exactly which NiceRF module are you using ?

chegewara, 2020-07-18:
NiceRF v1.0.
This is all i know. its small module with no label on top.

StuartsProjects, 2020-07-18:
Can you provide a link to the actual v1.0 module you bought and are using, NiceRF have several modules that are v1.0.

chegewara, 2020-07-18:
Ok, i think ive found it.
With SX1262 i have the same problem and i dont know how to explain it, but problem is `Serial.begin(115200);` with 9600, as in example, it is working.

chegewara, 2020-07-18:
I dont have link to sx1280 module, but i have working sx1262 module. DIO1 is switching and code seems to work. Now i have to connect 2nd module as receiver and test.

Thanks

chegewara, 2020-07-18:
With sx1262 RX is same problem, no IRQ on DIO1 even no timeout IRQ:

This is log when no IRQ wait:
```
13:46:21.277 -> 104_LoRa_Receiver_Detailed_Setup_ESP32 Starting
13:46:21.277 -> 
13:46:21.310 -> LoRa Device found
13:46:22.834 -> 
13:46:22.834 -> SX1262,915000000hz,SF7,BW125000,CR4:5,LDRO_Off,SyncWord_0x1424,IQNormal,Preamble_8
13:46:22.834 -> SX1262,PacketMode_LoRa,Explicit,LNAgain_Boosted
13:46:22.867 -> 
13:46:22.867 -> Reg    0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
13:46:22.867 -> 0x900  30 00 00 00 00 64 00 00 00 00 00 00 24 04 47 04 
13:46:22.867 -> 0x910  00 2F 00 00 00 03 0A 00 15 35 09 00 02 2B 70 08 
13:46:22.867 -> 0x920  07 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.867 -> 0x930  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.867 -> 0x940  00 07 00 03 02 00 10 00 0A 00 03 04 00 14 0C 00 
13:46:22.867 -> 0x950  00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.900 -> 0x960  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.900 -> 0x970  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.900 -> 0x980  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.900 -> 0x990  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.900 -> 0x9A0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.900 -> 0x9B0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.900 -> 0x9C0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.933 -> 0x9D0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.933 -> 0x9E0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.933 -> 0x9F0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:46:22.933 -> 
13:46:22.933 -> 
13:46:22.933 -> Receiver ready - RXBUFFER_SIZE 32
13:46:22.933 -> 
13:46:22.933 -> 2s PacketError,RSSI,-3dBm,SNR,-30dB,Length,0,Packets,0,Errors,1,IRQreg,0
13:46:23.198 -> 3s PacketError,RSSI,-3dBm,SNR,-30dB,Length,0,Packets,0,Errors,2,IRQreg,0
13:46:23.430 -> 3s PacketError,RSSI,-3dBm,SNR,-30dB,Length,0,Packets,0,Errors,3,IRQreg,0
```

chegewara, 2020-07-18:
I can confirm that sx1262 works with esp32 S2, both, transmit and receive examples.

Now the only problem is with sx1280.

Thanks

chegewara, 2020-07-19:
> Can you provide a link to the actual v1.0 module you bought and are using, NiceRF have several modules that are v1.0.

I believe this is the one i have:
https://www.nicerf.com/product_193_302.html

StuartsProjects, 2020-07-20:
'Believe' ? You dont sound sure. 

Where did the module come from are the pin labels on the back of the board (on the link) an exact match for the labels on your modules ?

Did you alter the original example program in anyway at all ?

chegewara, 2020-07-20:
Im sure its the one. Just label on bottom is slightly different, no 0/1 to check, on sx1280.

For first test i didnt change example code except pins. I tested all 3 begin() options and like i said example is working for sx1262, which is odd its not working for sx1280, because library code looks ok to me. 

One thing that may be interesting, but it may be normal, is transceiver code printing registers  0x00-0x4F is all 0x0s.

StuartsProjects, 2020-07-20:
The register print is of no consequence.

There must be some issue with your particular modules are the way they are connected.

chegewara, 2020-07-20:
Ok, thanks.
I dont think i messed soldering both sx1280, but i ordered today different type sx1280 modules from mouser. I will let you know the results.

Thanks

chegewara, 2020-07-22:
Hi,
i have new modules with sx1280 and it seems to work on esp32 S2 with stock examples. I only changed pins.
It was issue with other modules or with my breadboards.

Thanks

closed 2020-07-22 by chegewara