StuartsProjects/SX12XX-LoRa #17: Packet size and rate

yo90bosses, 2020-08-25:
I'm working on a telemetry system for projects based on the sx1280 and lora and need to know the max size of the packets and the rate I can send them at.
I understand it depends on the spreading factor, coding rate and such but Is there some kind of a way to calculate the max transfer rate of packets and their size or a clear explanation on a site to calculate this for the sx1280?
I'm probably going to use mavlink to create and manage the packets and some projects require at least 100 packets per second.

StuartsProjects, 2020-08-26:
The maxium size of a packet is in general 255 bytes, but depends on the mode you are transmitting in, the details are in the SX1280 data sheet. 

The air time of a packet, depending on its parameters, can be found using the Semtech calculator tool for SX1280.

yo90bosses, 2020-09-02:
Thanks for the information.
A few more questions have come up though.
What settings need to be the same between modules for them to work and if one module transmits a message do all other modules with the same settings receive it or only one?

StuartsProjects, 2020-09-02:
See the example program;

103_LoRa_Transmitter_Detailed_Setup

Which contains the detail of the setup for the LoRa device;

`//***************************************************************************************************
  //Setup LoRa device
  //***************************************************************************************************
  //LT.setMode(MODE_STDBY_RC);
  //LT.setRegulatorMode(USE_LDO);
  LT.setPacketType(PACKET_TYPE_LORA);
  LT.setRfFrequency(Frequency, Offset);
  //LT.setBufferBaseAddress(0, 0);
  LT.setModulationParams(SpreadingFactor, Bandwidth, CodeRate);
  LT.setPacketParams(12, LORA_PACKET_VARIABLE_LENGTH, 255, LORA_CRC_ON, LORA_IQ_NORMAL, 0, 0);
  //LT.setDioIrqParams(IRQ_RADIO_ALL, (IRQ_TX_DONE + IRQ_RX_TX_TIMEOUT), 0, 0);
  //***************************************************************************************************
`
I have commented out the settings  that dont need to match. 

And yes, then if the settings for the receivers match those for the transmitter then all the receivers (in range) will receive the packet, how could it be any other way .................

yo90bosses, 2020-09-04:
I'll be using an ESP32 and it will go into deep sleep but the sx1280 will be receiving packets from other nodes.
Will the packet be lost when the ESP32 boots and calls the setup function again? Or can I skip the setup function when it boots?

StuartsProjects, 2020-09-04:
The ESP32 is not an ideal processor for such an application. 

In particular the so called 'deep sleep' mode does odd things with the IO pins both going into sleep and coming out of it, either of these can interfere with the setup stored in the LoRa device. 

You would need to write your own routines to make a difference between a first time reset and each subsequent reset. 

In particular the library ::begin() function, run at program start\reset does reset the LoRa device.  

There is an example that demonstrates wake up from 'deep sleep' on reception of a LoRa packet, but its designed, like most of the library, to run on an ATmega processor.

yo90bosses, 2020-09-12:
I want to receive and send packets at the same time. Currently im calling .receive(RXBUFFER, 255, 0, NO_WAIT) and checking .isReceiveDone() to se if a packet has arrived. But if i want to send a packet i call .transmitSXBuffer(0, TXPacketL, 4000, TXpower, WAIT_TX) after writing to the buffer. Both the transmitting part and the receiving parts work fine on their own but if i do both, when the program send a packet it fails and then receives right after that an incorrect packet.

Im basically trying to receive a packet from one node and then sending a packet back to the node. It cant be blocking as it has to constantly be doing other things.

Am I doing something wrong or is this not possible?

StuartsProjects, 2020-09-13:
If you transmit a packet is destroys the receive settings in the device. The only way to then receive a packet it to use .receive again.

yo90bosses, 2020-09-17:
In the datasheet it says the current consumption when receiving it around 6 mA. Is this only when receiving a packet or also when it is waiting for a packet?

StuartsProjects, 2020-09-17:
No idea, but easy enough to measure.

closed 2020-09-17 by StuartsProjects