StuartsProjects/SX12XX-LoRa #18: SX1261, setMode function ...

MC123T, 2020-08-28:
can someone explain the setMode function?
The SX1261 datasheet (page60/111) gives a illustration wich function is needed to set the chip in the wanded mode ..
like setSleep() (wich is implemented in the libary..) setMode() is not mentioned somewhere
are the functions setRx(), setTx() and setFs() missing ? SetMode() is using opcode  RADIO_SET_STANDBY (0x80) wich is the right opcode for setStandy()..

Should setMode() be adjusted to setStandy? i dont get the purpose of setMode();
How is the chip set to Tx/RX ? there is no function call like setMode(MODE_TX) ?

StuartsProjects, 2020-08-28:
First, appreciate that the objective of the library was to maintain a relativly constant library structure and applications sketch format across the range of LoRa devices, SX127x, SX128x and SX126x. These devices are however configured very differently. 

In the context of the SX126x part of the library then SetMode() is indeed redundant, but its there so that the library code for SX127x (and to a lesser extent the SX128x) is very similar to that for the SX126x. This similarity has made porting library functions across LoRa module types very much easier. In an early version of the SX126x library there was no SetMode(), it was then added for compatibility reasons. 

setRx() and setTX() functions are implemented.

MC123T, 2020-08-28:
alright, that makes sence to me! sorry, i didnt took a look at the ther chips.

sx1261lt.h includes a list of radio modes : 

> #define MODE_SLEEP                                  0x07   														
#define MODE_STDBY_RC                               0x00   //Device running on RC13M, set STDBY_RC mode 		
#define MODE_STDBY_XOSC                             0x01   //Device running on XTAL 32MHz, set STDBY_XOSC mode
#define MODE_FS                                     0x02   // The radio is in frequency synthesis mode
#define MODE_TX                                     0x03   //TX mode
#define MODE_RX                                     0x04   //RX mode
#define MODE_RX_DC                                  0x05   //RX duty cycle mode
#define MODE_CAD                                    0x06   //RX CAD mode

they are used as modeconfigs in setMode();
the Datasheet only lists 2 of them: 0x01(STDBY_XOSC) and 0x00(STDBY_RC).
I cant find the other ones in the datasheet. I assume they can be used on different chips.

you may want to delete those lines from the header file. correct me if iam wrong..
this confused me because i thought, i could use these operating modes with setMode() (wich is not possible, right?)

this thread can be closed then :)

closed 2020-09-17 by StuartsProjects