StuartsProjects/SX12XX-LoRa #19: Redefine SPI pins

ddosoff, 2020-10-11:
Hello Stuart,

I want to connect TTGO-T-Display ESP32 board with EBYTE E28 2G4M20s radio.

Using T-Display board I have to reassign SPI pins. I see that you use predefined SPI object directly.

Is it possible to reassign SPI pins on the basic SPI object?
Or I have to modify your library to accept SPIClass on the initialization state?

ddosoff, 2020-10-11:
I found that simple: 

SPI.begin(E28_SCK, E28_MISO, E28_MOSI, E28_NSS);

did the works. Sorry for disturb. And thanks for your work.

closed 2020-10-11 by ddosoff

StuartsProjects, 2020-10-11:
Note that information on re-assigning the SPI pins on ESP32 is in the readme;

"Alternatively you can uses this format to start SPI;
SPI.begin(SCK, MISO, MOSI, NSS);
"

ddosoff, 2020-10-12:
Finally, I got working TTGO T-Display + EBYTE E28 2G4M20S. I believe this message will be googled by someone and this helps him to build his project faster.

For EBYTE SX1280 module you **should attach RX_EN and TX_EN lines**. This knowledge takes about 1 day from my life.

My pinout:

```
#define E28_MISO 27
#define E28_MOSI 26
#define E28_SCK 25
#define E28_NSS 33
#define E28_DIO1 17
#define E28_BUSY 22
#define E28_RESET 21
#define E28_TXEN 15
#define E28_RXEN 38
```

Initialization:

```
SPI.begin(E28_SCK, E28_MISO, E28_MOSI, E28_NSS);

if (LT.begin(E28_NSS, E28_RESET, E28_BUSY, E28_DIO1, -1, -1, E28_RXEN, E28_TXEN, DEVICE_SX1280))
    Serial.println(F("SX1280 device found"));
else
{
    Serial.println(F("SX1280 not responding"));
    while (1)
      ;
}

LT.setupLoRa(Frequency, Offset, SpreadingFactor, Bandwidth, CodeRate);
```

happytm, 2021-07-25:
@ddosoff  Thank you for providing pinout and initialization code for Ebyte module. This will be time saver for lot of people using these modules.

Thanks.