StuartsProjects/SX12XX-LoRa #20: Non blocking receiver

ddosoff, 2020-10-12:
Hello Stuart,

I try to do the best understanding how to correctly implement non blocking receiver. Please give a tip.

StuartsProjects, 2020-10-13:
There are several examples demonstrating this. 

25_GPS_Tracker_Receiver_With_Display_and_GPS for instance shows the program reading a GPS whilst waiting for a packet to arrive.

ddosoff, 2020-10-14:
In the example you call _receiveSXBuffer(0, 0, NO_WAIT)_ each loop. I tried to call this function only once before receiving packet.

I found that is necessary to call

```LT.receiveSXBuffer(0, 0, NO_WAIT);```

again after reading buffer with 

```
LT.startReadSXBuffer(0);
...
LT.endReadSXBuffer();
```

to make it clear and ready to receive next packet.
Is it correct way?

StuartsProjects, 2020-10-14:
> Without calling _receiveSXBuffer(0, 0, NO_WAIT)_ SX1280 can't receive the next packet.
> 
> Is it correct way?

That is correct and the method used in the examples.

closed 2020-10-15 by ddosoff