StuartsProjects/SX12XX-LoRa #21: PA_BOOST?

CongducPham, 2020-10-13:
Hi, first of all, great job! Then, I didn't see how PA_BOOST can be selected in the lib. Is there a way for that? regards,

StuartsProjects, 2020-10-13:
There is no sepearte function to enable or disable PA_BOOST. 

If you follow the code for one of the transmit functions you will see that PA_BOOST is applied automatically if power greater than 17dBm is requested.

CongducPham, 2020-10-13:
Hi, I went through SX127XLT.cpp and I found this portion of code:

  if (_Device != DEVICE_SX1272)
  {
    //for all devices apart from SX1272

    if (txPower > 17)
    {
      param1 = (txPower + 0xEB);
      param2 = OCP_TRIM_130MA;
    }

    if (txPower <= 17)
    {
      param1 = txPower + 0xEE;
      param2 = OCP_TRIM_100MA;
    }

    if (txPower <= 5)
    {
      param1 = txPower + 0xEE;
      param2 = OCP_TRIM_80MA;
    }

  }
  else
  {
    //for device SX1272

    if (txPower > 17)
    {
      param1 = (txPower + 0x7F);
      param2 = OCP_TRIM_130MA;
    }

    if (txPower <= 17)
    {
      param1 = txPower + 0x82;
      param2 = OCP_TRIM_100MA;
    }

    if (txPower <= 5)
    {
      param1 = txPower + 0x82;
      param2 = OCP_TRIM_80MA;
    }

  }

  writeRegister(REG_PACONFIG, param1);       //Reg 0x09 this changes for SX1272 and SX1278
  writeRegister(REG_OCP, param2);            //Reg 0x0C this is same for SX1272 and SX1278

From what I can see, it seems that radio others than 1272 have PA_BOOST regardless of requested power. Am I correct or am I missing something?

StuartsProjects, 2020-10-13:
> From what I can see, it seems that radio others than 1272 have PA_BOOST regardless of requested power. Am I correct or am I missing something?

I do not understand the point you are making, could you explain what you are actually trying to do ? 

That part of the library is for the SX127x devices and all of the devices in that range are programmed the same, apart from the SX1272.

CongducPham, 2020-10-13:
Well, some manufacturers use the 1276 chips but do not use the PA_BOOST line, so in that case, using DEVICE_SX1276 will force PA_BOOST to be used while the circuit can be disconnected because the RFO line is used instead.

StuartsProjects, 2020-10-13:
PA_BOOST is only enabled if you set the power to more than 17dBm. 

If you have an SX1276 module that does not have PA_BOOST connected up, you should not be setting the power to more than 17dBm.

CongducPham, 2020-10-14:
Hi, one issue I can see is for instance the RFM95W which is based on SX1276 but using only PA_BOOST line. If I set the power to 14dBm for Europe regulation then PA_BOOST is not activated according to your setting. But in doing so, the output signal will be very weak (negative SNR even at 3m) because the RFO line is not physically connected.

CongducPham, 2020-10-14:
Hi, unfortunately, it seems that HopeRF just took the Semtech documentation and changes SX1276 by RFM95! NiceRF modules and Modtronic inAir9B modules do the same, only PA_BOOST line is wired. For RFM95W there are a number tests showing that only PA_BOOST line is wired (https://www.disk91.com/2019/technology/lora/hoperf-rfm95-and-arduino-a-low-cost-lorawan-solution/ for instance). The reason it is working for you is that in your code, it seems that for radio others than 1272 and, regardless of requested power, PA_BOOST is used : bit 7 of RegPaConfig is always set. If you remove bit 7 and use output power less than 17dBm with an RFM95 you will get very weak signal because RFO line is not wired.

StuartsProjects, 2020-10-15:
I see. 

So the library code works with the RFM95, and other modules, but wont work if a module is using only the RFO outputs.  

Do you have or do you know of a module using the SX1276/7/8/9 that use the RFO output, most I have seen use PA_BOOST ?

CongducPham, 2020-10-15:
The Modtronic inAir9 (not the inAir9B) is based on SX1276 and uses only the RFO line.

CongducPham, 2020-10-15:
Hi, BTW, it seems that the IQ inversion is not correct. The latest version of SX1276 reference indicates that IQ inversion should take into account whether it is for TX (bit 0 of reg 0x33 set and bit 6 of reg 0x33 cleared) or for RX (bit 6 of reg 0x33 set and bit 0 of reg 0x33 cleared). We did have a lot of issues on this so this is why we found that setting.

StuartsProjects, 2020-10-15:
Yes, I know about that.  It works for the normal setting but not inverted. 

There are a few changes to the SX127x library pending including adding some new features etc, but I need to go through and check most of the examples against the changes first.

StuartsProjects, 2020-10-15:
> The Modtronic inAir9 (not the inAir9B) is based on SX1276 and uses only the RFO line.

I will add a note that those modules are not supported. 

Unfortunatly I dont have any modules that use only the RFO line and I wont make changes to the library unless I am able to test them.

CongducPham, 2020-10-15:
Hi, would you consider adding a function to independently control the PA_BOOST? Default behavior would be your current code and then for specific modules, user can call an additional function to set PA_BOOST options.

StuartsProjects, 2020-10-15:
> Hi, would you consider adding a function to independently control the PA_BOOST? Default behavior would be your current code and then for specific modules, user can call an additional function to set PA_BOOST options

I would consider it, however its a significant cost to me to get hold of appropriate modules to do the testing.

StuartsProjects, 2020-10-15:
> I have quite a lot of these inAir9 I can send you one.

There is a contact page on the website;

[](https://stuartsprojects.github.io/)

closed 2022-07-14 by StuartsProjects