StuartsProjects/SX12XX-LoRa #22: RSSI and SNR should be signed

ddosoff, 2020-10-15:
Hello Stuart, 

I found that RSSI and SNR values should be signed.
And you have signed types here: https://github.com/StuartsProjects/SX12XX-LoRa/blob/master/src/SX128XLT.h#L172

But functions to get RSSI and SNR declared with unsigned return values: https://github.com/StuartsProjects/SX12XX-LoRa/blob/master/src/SX128XLT.h#L82

One more thing is correct sign conversion. Functions correcly convert received value working with signed bytes.

Something like:

```
  int8_t status[5];

  readCommand(RADIO_GET_PACKETSTATUS, (uint8_t *)status, 5);
  _PacketRSSI = -status[0] / 2;
```

status should be int8_t, not uint8_t to correctly calculate RSSI.

StuartsProjects, 2020-10-15:
Does any of the example code incorrectly display the RSSI or SNR ?

ddosoff, 2020-10-15:
In my application - yes. Value jumps like 1,2,3,127,128..

StuartsProjects, 2020-10-15:
I need to see the code, to compare it against the working examples.

ddosoff, 2020-10-15:
I did just dirty fix:

main.cpp:
```
int8_t PacketRSSI = LT.readPacketRSSI();
snprintf(buf, sizeof(buf), "%ddBm", (int)PacketRSSI);
```

sx128XLT.cpp:
```
uint8_t SX128XLT::readPacketRSSI()
{
#ifdef SX128XDEBUG
  Serial.println(F("readPacketRSSI()"));
#endif

  int8_t status[5];

  readCommand(RADIO_GET_PACKETSTATUS, (uint8_t *)status, 5);
  _PacketRSSI = -status[0] / 2;

  return _PacketRSSI;
}
```

To make it clear, return value of readPacketRSSI should be integer.
For example, like here: https://os.mbed.com/teams/Semtech/code/SX1280Lib/file/abf14b677777/sx1280.cpp/ (line of code 455).

Also it is possible shorter SPI read with ReadCommand( **RADIO_GET_RSSIINST**, &raw, 1 ).

StuartsProjects, 2020-10-15:
So with your fix, what sort of RSSI are you seeing reported when TX and RX are about 1m apart ?

StuartsProjects, 2020-10-15:
Can you also indicate what suggted to you that the RSSI and SNR values actually reported by the example programs were not correct ?

CongducPham, 2020-10-17:
Hi, I was also looking at this issue and I believe the way the packet RSSI is computed also depend on SX1272 (-139+PacketRSSI) or SX1276 (-157+PacketRSSI). Also, we should take into account low frequency (-164+PacketRSSI) such as 433MHz band. Also, it is advised to take into account the SNR and see whether it is positive or negative to correct the value of PacketRSSI. See 3.5.5 (page 87) of SX1276  spec (May 2020).

StuartsProjects, 2020-10-17:
Semtech have changed (in the data sheets) how RSSI and SNR are calculated a number of times. 

Whether software libraries should change every time Semtech revise the RSSI calculation is a moot point, there is a strong argument for leaving the calculation alone so that firmware from earlier versions operates in the same way as newer versions. 

And as Semtech themselves point out if you want RSSI to be accurate then you need to calibrarte it for each setup.

StuartsProjects, 2020-10-24:
RSSI has a value of between 0 and -127dBm. There is no indication in the data sheet I can see that a positive value is possible. 

There is no report of any example program reporting the wrong values for RSSI.

closed 2020-10-24 by StuartsProjects