StuartsProjects/SX12XX-LoRa #23: _TXPacketL as int8_t?

CongducPham, 2020-10-17:
Hi I see that _TXPacketL is defined as int8_t. Shoudn't it be defined as uint8_t? I'm not sure how it will behave when assigning _TXPacketL = size when size is greater than 128. It will probably wrap to negative value. Then when readTXPacketL() returns _TXPacketL as uint8_t maybe it will give back the correct value. So is this an error or int8_t was chosen on purpose?

StuartsProjects, 2020-10-17:
The functions where _TXPacketL is used operate correctly, it could be defined as uint8_t, but it makes no difference to the operation of the code as it is that I am aware of.   

Can you point an example program that does not operate correctly ?

CongducPham, 2020-10-17:
I didn't try an example, just wondering why int8_t when going through the code. Especially because _RXPacketL is defined as uint8_t.

StuartsProjects, 2023-02-13:
No further action needed

closed 2023-02-13 by StuartsProjects