StuartsProjects/SX12XX-LoRa #24: SX1280 LT.receive and ranging slave problems

VILLO88, 2020-10-23:
Hi Stuart, many thanks for this incredible library!
I am testing the rangine function of the sx1280 and I am very satisfied with the accuracy, but I have a problem: in my application there are several nodes that regularly wake up and enter LT.receive first to check the call address and then enter ranging slave , the others have to go back to sleep.
The problem is that if the base station is in ranging master and a node enters LT.receive, the function gets stuck at this while line 

uint8_t SX128XLT::receive(uint8_t *rxbuffer, uint8_t size, uint16_t timeout, uint8_t wait)
{
#ifdef SX128XDEBUG
  Serial.println(F("receive()"));
#endif

  uint8_t index, RXstart, RXend;
  uint16_t regdata;
  uint8_t buffer[2];

  setDioIrqParams(IRQ_RADIO_ALL , (IRQ_RX_DONE + IRQ_RX_TX_TIMEOUT), 0, 0);  //set for IRQ on RX done or timeout
  setRx(timeout);

  if (!wait)
  {
    return 0;                                                           
  }
  while (!digitalRead(_RXDonePin));                                         //GET STUCK HERE !!!
  setMode(MODE_STDBY_RC);                   
....
..

Any ideas?  thanks

StuartsProjects, 2020-10-24:
If a node is in packet receive mode, it is not 'stuck', it is waiting for a valid packet. 

If you want the node to receive a ranging request the node needs to be in receiveRanging() mode.

VILLO88, 2020-10-24:
Yes i know that, i explain better: i have several nodes that periodically wake up and listening for normal LoRa type packet containing an address identification request or other info this doesent matter, and they get stuck in LT.receive if there is a ranging comunication between other 2 modules.
It get stuck waiting for the DIO1 interrupt pin go high if an IRQ_RX_DONE or IRQ_RX_TX_TIMEOUT happens, but why?

StuartsProjects, 2020-10-24:
Like I said its not 'stuck' its waiting for a packet to arrive. 

Presumably you have called LT.receive with WAIT set. 

For ways of using LT.receive without forcing the wait, read the notes on the main Github page for the library.

VILLO88, 2020-10-24:
ok thanks

closed 2020-10-29 by StuartsProjects