StuartsProjects/SX12XX-LoRa #25: 2 minor typos in SX128XLT.cpp

CongducPham, 2020-10-26:
Hi, going through SX128XLT.cpp I believe there are 2 minor typos you may want to correct for next commit:

line 1320: https://github.com/StuartsProjects/SX12XX-LoRa/blob/02c237007cc7259445c90f7c1d33aa3d0b04cac7/src/SX128XLT.cpp#L1320. The F() is probably missing.

line 1927: https://github.com/StuartsProjects/SX12XX-LoRa/blob/02c237007cc7259445c90f7c1d33aa3d0b04cac7/src/SX128XLT.cpp#L1927. It is probably #ifdef SX128XDEBUG1 instead of #ifdef SX127XDEBUG1

Again, great job. I will start testing compiling the same code for both SX127X and SX128X radio modules
regards,

CongducPham, 2020-10-26:
line 2293: https://github.com/StuartsProjects/SX12XX-LoRa/blob/02c237007cc7259445c90f7c1d33aa3d0b04cac7/src/SX128XLT.cpp#L2293. It is probably #ifdef SX128XDEBUG1 instead of #ifdef SX126XDEBUG1

StuartsProjects, 2020-10-29:
Thank you, and no doubt there are other typos to be found.

StuartsProjects, 2021-11-14:
There has recently been a number of changes to the SX128X library, specifically to add 'relaible' packet support. At the same time the typos were corrected.

StuartsProjects, 2021-11-14:
Corrected

closed 2021-11-14 by StuartsProjects