StuartsProjects/SX12XX-LoRa #27: Troubles with using remote control examples

delchrys, 2020-10-28:
Hi,
Good work on the library and examples. 
I'm having problems when using the remote control examples. 
I have two Arduino Unos. 
Two adafruit 3073 Lora modules. Rfm96w or sx1276.
All examples work correct like the detailed setup and basic programs, both sides tx/rx. 
But when I try the remote control examples my TX with debug works fine. It sends: 101,X1,123,Y1,128,11111111,3ms

But the receiver loops through setup. Says 36_remote_control_servo_receiver ready.

Err, -97dBm

Indont know what the problem might be. 
Is it possible you can help me with this because I really like and need this example to work.

StuartsProjects, 2020-10-29:
I made some changes to the examples, they were working OK at spreading factor 7, but not when swithed across to spreading factor 6 where a change to implicit mode was required.

delchrys, 2020-10-29:
Great. I've been thinking for two days what the difference might be. I had some options which I was going to try today, among those options was the spreading factor. I will try the change in about 3h when I'm at work (where the project is).
I'll let you know

StuartsProjects, 2022-12-06:
No issues with library

closed 2022-12-06 by StuartsProjects