StuartsProjects/SX12XX-LoRa #28: SX127X issue when receiving long messages?

CongducPham, 2020-11-09:
Hi, it seems that when using rxtimeout!=0 there can be an issue to receive long message (at least the probability of an issue is higher). That happens when you start receiving a message at time t and the complete message would be entirely received at time t+a. If t+a > endtimeoutmS then you will timeout. For instance in your code: https://github.com/StuartsProjects/SX12XX-LoRa/blob/7e912530344dc7b5d0b0fe0c4281047366b1d874/src/SX127XLT.cpp#L1906

I don't know if this is the intended behavior. Maybe, as rxtimeout can be considered as a firm deadline. In my case, however, since I need to use an rxtimeout !=0 to be able to get out of the receive function periodically, I check for a valid header and once got a valid header I will just extend the rxtimeout to allow for a complete reception of the packet.

This issue is only for SX127X as SX126X and SX128X have the rx timeout handled nicely by the chip itself: "As soon as a packet is detected, the timer is automatically disabled to allow complete reception of the packet." from Semtech specs.

Hope that helps.
Regards,

StuartsProjects, 2020-11-09:
That is the intended behaviour of the timeout that was added to the SX127x.

trlink, 2023-12-04:
In my case this results to a reboot because the watchdog got triggered,,,

StuartsProjects, 2023-12-05:
The SX127X does not have the internal functions to instigate a timeout. The timeout function was added to the SX127X library to make routines compatible with the SX126X and SX128X sketches. There is no plan to modify the timeout functionality of the library.

closed 2023-12-05 by StuartsProjects