StuartsProjects/SX12XX-LoRa #29: Possible reasons for ERROR - Busy Timeout! message on SX1280

yo90bosses, 2020-11-19:
Im using the SX1280 (Ebyte 2G4M20S) and as far as i understand when i try to enter sleep mode i get the message ERROR - Busy Timeout! on the serial port and it still draws too much current to be in sleep mode. Because the program is way to big to post i just want to know what could be possible causes for this message to help find out what is causing it.

Thanks in advance.

StuartsProjects, 2020-11-19:
Some questions;

Do the example transmitter and receiver programs (103 and 104) work correctly with your hardware setup ?

Do you get the Busy Timeout error with program '5_LoRa_TX_Sleep_Timed_Wakeup_Atmel' ?

What is the actual sleep current when using the program  '5_LoRa_TX_Sleep_Timed_Wakeup_Atmel' ?

yo90bosses, 2020-11-20:
Im using an ESP32 and my code that controls the sx1280 worked when i first wrote it but after integrating it into a big project (In which it worked at first) and doing some big changes, this error started. I thought because the library is giving me this message you could shed some light on what some possible errors could be.

When i was testing the board, yes all the transmitter and receivers programs worked. As to the sleep, that worked as well and the receiver did sleep. Actual sleep current i couldnt directly measure as it was all integrated into one board but when sleeping it was under 1ma and when in receive mode 8ma.

This almost always happens when .setSleep(CONFIGURATION_RETENTION); is called. But sometimes (rarely) also somewhere else in my code for the sx1280.

Does this error message come from multiple things or is there only one possible cause for it?

yo90bosses, 2020-11-20:
Ive found that the other times the error message comes, is when .transmit(_buffer, length, 0, power, NO_WAIT); is called.

StuartsProjects, 2020-11-20:
First off, the ESP32 is not directly supported, as mentioned in the readme. The .setSleep code and sleep mode has not beed tested against an ESP32 and there are no examples of sleep with an SX1280 and ESP32 in the library. 

The ESP32 can be difficult to use with deep sleep, very odd things happen with the IO pins going into an out of sleep and it would be impractical to customise the library to support the ESP32 in these circumstances. It can take a while to work out a stable and reliable use of sleep with the ESP32.

The busy timeout occurs when the LoRa device stops responding over SPI, or BUSY pin cannot be read for some reason. 

All I can suggest is that you reduce the program to a minimal example that exhibits the problem with your setup.

StuartsProjects, 2020-11-20:
You need to be checking that the example programs are working correctly, do you ever see busy timeout with any of the library examples ?

If you get a busy timeout when .transmit is called, it may be that transmit is being called when a previous transmit is still in progress.

StuartsProjects, 2020-12-21:
No issue with library, connection error.

closed 2020-12-21 by StuartsProjects