StuartsProjects/SX12XX-LoRa #30: SX1280 very low range

axelJ31, 2020-12-19:
I'm using the E28-2G4M20S with the pcb antenna and an arduino pro mini. It's powered using a 800mA 3.3V regulator.
The issue is that I can receive the data only when the two modules are 1 mm close. 
Thanks for your help.

StuartsProjects, 2020-12-19:
What pins do you have connected between the E28-2G4M20S and the Pro Mini ?

axelJ31, 2020-12-20:
Thanks for your reply,
I have connected : NSS to pin 10
                              NRESET to pin 3 
                              RFBUSY to pin 9
                              DIO1 to pin 2
                              SCK to pin 13
                              MISO to pin 12
                              and MOSI to pin 11

StuartsProjects, 2020-12-20:
Could be worth reading the readme section 'Considerations for pin usage' on the first page of this repository. 

Do any of the comments there apply to your module ?

axelJ31, 2020-12-20:
The TX and RX pins of my Ebyte module aren't connected. Are they necessary ?

StuartsProjects, 2020-12-20:
I  cannot think, myself, why there would be RX and TX pins provided if they were not needed. 

Perhaps check the datasheet for your modules ?

axelJ31, 2020-12-20:
After connecting the TX and RX pins, it works !
Thank you.

closed 2020-12-20 by axelJ31