StuartsProjects/SX12XX-LoRa #34: SX128x - sample range RSSI & EFI?

maholli, 2021-04-03:
Hi Stuart! 

Love the work you've done on the SX1280 + ranging. I use the radio quite a bit for my [V-R3x cubesat mission](https://www.nasa.gov/ames/v-r3x/) and have a hacky 1280 library in CircuitPython as well. The care and effort you've put into your libs is apparent and appreciated! 

Anyways, I noticed you don't sample the ranging rssi while reading the ranging registers. Any reason why? It's the next register after the ranging values (`0x0964`). The register isn't identified in the datasheet, but it's present in Semtech's C++ lib as `REG_RANGING_RSSI`.

Additionally, have you played around much with measuring the frequency error between the host and the ranging node prior immediately prior to performing a ranging exchange? I've found this quite helpful in fine-tuning the distance value for cm accuracy.

StuartsProjects, 2021-04-04:
I dont remember the specifics, but I do recall trying the standard RSSI read and found that did not work, I guess I was paying too much attention to the datasheet, and did not come across  REG_RANGING_RSSI. Semtech have never released the full details of what all those registers are, same situation for the SX126X.

I can see there are potential benefits in adjusting the Master and Slave to be closer together in frequency the frequencies should be fairly stable in the very brief interval that the ranging process takes, yet there is still a considerable variation in distance amounting to 5M or more between readings.

I dont quite see how improving precision in itself would help reduce that instability, unless you are saying your getting a much more stable reading ? What variation at say 50M are you getting ?

There is a TCXO equipped SX1280 out there that should improve the distance stability but I dont have one.

StuartsProjects, 2021-04-04:
Well good luck with that, whens the launch ?

As you may know I worked with Professor Twiggs quite a bit in around 2012 on the PocketQubes. A couple of years back I almost got another free Launch and would have loved to do a very simple UHF LoRa PocketQube.

maholli, 2021-04-04:
> I dont remember the specifics, but I do recall trying the standard RSSI read and found that did not work, I guess I was paying too much attention to the datasheet, and did not come across [...]

Yes I've found packet and instantaneous RSSI after ranging is explicitly wrong. However, reading out 4 bytes instead of 3 when retrieving the ranging value is supposed to contain the ranging RSSI as the last byte. It's not clear to me if this byte should be treated like the packet RSSI byte (where you divide by 2 and consider it negative). Refer to Semtech's `GetRangingPowerDeltaThresholdIndicator` function in their [mbed library](https://os.mbed.com/teams/Semtech/code/SX1280Lib//file/c4f110f3fe3e/sx1280.cpp/).

> I dont quite see how improving precision in itself would help reduce that instability, unless you are saying your getting a much more stable reading ? What variation at say 50M are you getting ?

What I was referring to is the use of the "error frequency indicator" to adjust the ranging measurement according to a linear correction outlined in Semtech's ranging app note. Note I've found the table in excerpt below is very specific to the signal path they characterized on their dev boards. It's necessary to generate your own correction values when using the niceRF modules, for example. 

[image]
(from: "Application Note: An Introduction to Ranging with the SX1280 Transceiver")

StuartsProjects, 2021-11-14:
There has been a fair few changes to the SX128X library recently, primarily to add 'reliable' packet and data transfer functions. 

At the same time the SX128X RSSI code was amended and a getRangingRSSI() function added, example 54_Ranging_Master now includes a printout of the ranging RSSI.

closed 2022-12-06 by StuartsProjects