StuartsProjects/SX12XX-LoRa #36: Channel Activity Detection (CAD) and Listen Before Talk (LBT) implementation

mamanbudiman, 2021-05-25:
I have implemented this library on STM32L052C8 with SX1276 Lora module without problem.

Is the any hint to implement the CAD and LBT using this LoRa library ?

Best regards, 
Maman

StuartsProjects, 2021-05-26:
I suspect that all you need to do is set the CAD IRQ then turn on RX with a timeout. 

Review the code in one of the library :receive functions for an example.

mamanbudiman, 2021-05-26:
I want to transmit a packet data, but before transmitting the data, I want to check whether the channel is free or not.
If it is not free, I have to wait till the channel is free.

StuartsProjects, 2021-05-26:
Then you will need to write a function that replicates the :receive function code but has the CAD detect IRQs set, so when the function times out you can test if the timeout was caused by CAD. 

I dont have a tested working example of such a function.

StuartsProjects, 2023-02-13:
User will need to write a CAD detect function, none exists in the current library.

closed 2023-02-13 by StuartsProjects