StuartsProjects/SX12XX-LoRa #37: Guidence needed for Audio link using SX1280 chip.

happytm, 2021-07-01:
Mr. Robinson,

Thank you for your excellent library and your documentation of experiments you are doing with LoRa.

Your blog is my reference for learning LoRa technology. You make it very simple to understand for new people to LoRa like me.

I have ordered some SX1280 based modules from Ebyte linked below:

https://www.aliexpress.com/item/1005001499235376.html?spm=a2g0o.productlist.0.0.43721fe4HHBdn1&algo_pvid=657035e1-932c-4d93-a12c-5fba8edd1a9e&algo_exp_id=657035e1-932c-4d93-a12c-5fba8edd1a9e-2

I would like to use this modules for audio link using ESP32 (Wemos lolin32) board pairing it with Ebyte SX1280 module linked above.

I found a walkie talkie project using ESP32 with SX1278 chip at link below:

https://github.com/deulis/ESP32_Codec2

Unfortunately , Lora library used in above project does not support SX1280 chip.

Is it possible to use your library to create project like above using SX1280 chip with some tweaks? If so can you guide me with some hints ?

I am eagerly waiting for your final code for ESP-Cam project too.

Thanks.

StuartsProjects, 2021-07-02:
> Is it possible to use your library to create project like above using SX1280 chip with some tweaks? If so can you guide me with some hints ?

Probably, but you might want to try and match the functionality with the SX127x part of the library first, there would be less to change and once working it should port easily enough across to SX128x. 

Probably the key thing to get right is to send a test packet with the original code and then be sure its read correctly with the SX127x library code.

happytm, 2021-07-02:
OK. Once I receive the modules I will give it a try.

Thanks.

StuartsProjects, 2022-12-06:
No issues with library