StuartsProjects/SX12XX-LoRa #38: Reference values for RSSI and SNR

maguimaraes, 2021-07-07:
Hi,

I'm testing LoRa GNiceRF SX1280 modules with Arduino Nano. I'm getting the following SNR and RSSI values when I touch the antennas (back to back). The following values are presented when communication is established:

_Hello World 1234567890*,CRC,DAAB,RSSI,-10dBm,SNR,14dB,Length,23,Packets,1,Errors,0,IRQreg,8012_

I would like to know what are the acceptable values of RSSI and SNR for this distance (=1cm).

StuartsProjects, 2021-07-07:
Bad idea to have a transmitting antenna so close to a receiving one, you risk overloading or damaging the receiver. 

As for whether values recorded are acceptable or even accurate, I do not know, its more a question for the module manufacturer or Semtech.

StuartsProjects, 2021-11-14:
Not a library issue

closed 2021-11-14 by StuartsProjects