StuartsProjects/SX12XX-LoRa #39: DIO1 pin remains HIGH after transmit/receive

ONLYstcm, 2021-08-04:
Doing some debugging and I noticed that DIO1 remains HIGH after a transmit or receive is completed. Looking into the source code, I couldn't seem to find where the pin is reset to LOW. Or a function that can be used to reset the pin. Is this something, that us developers should manually control in our code?

StuartsProjects, 2021-08-05:
And which LoRa device are you using, DIO1 pin use does vary across the module families ?

ONLYstcm, 2021-08-05:
I'm using the RFM99W by HopeRF

StuartsProjects, 2021-08-05:
Which LoRa device does that Hope module use ?

ONLYstcm, 2021-08-05:
SX1280

StuartsProjects, 2021-08-05:
DIO1 remains high after completion of RX and TX by design, there is library code to map DIO1 to the appropriate SX128X interrupt.  

DIO1 is reset at the beginning of the transmit() and receive() functions, so see the library code for the IRQ functions that do this.

ONLYstcm, 2021-08-05:
clearIRQ function worked

closed 2021-08-05 by ONLYstcm