StuartsProjects/SX12XX-LoRa #41: Tx Continuous wave no output

hpk-roquette, 2021-09-24:
Hi,

I'm having a hard time using the RADIO_SET_TXCONTINUOUSWAVE (opcode D1) on a SX128X. I am wondering if I'm missing an obvious step.

If I loop and transmit packets (SetTx), I am able to see them on the power meter, so I know that I transmit successfully.

For the continuous wave, I do the following:
  - set the RF frequency
  - set the TX Params (output power and ramp time)
  - set the CW with opcode D1. 
 
I see a small blip on a power meter, then nothing. Am I reading the documentation wrong? Is there a step missing?

Thanks!

StuartsProjects, 2021-09-25:
I did not copy\implement the carrier wave modes that had been developed in the similar SX126X Library and used for generating tones, FSK RTTY etc, across to the SX128X.

I remember considering it, but could not think of a good use for tones and the like on 2.4Ghz and neither did I have anything to use to listen to them.

Whats the application ?

hpk-roquette, 2021-09-27:
Thanks for the response,

The goal of the single tone was just for testing purposes, to check the power output in a power meter. When transmitting "normal" packets, I would only see around 0dBm on the power meter, and thought it was a bit weak. I was trying to figure out why, using a simpler continuous wave.

StuartsProjects, 2021-10-02:
I just check the power output with a LoRa packet lasting several seconds, the power output levels seem to be correct, TX output of  10dBm measures as 9.2dBm and -18dBm measures as -20dBm.

hpk-roquette, 2021-10-07:
We did manage to setup a single tone for our test setup. Thanks for the responsiveness !