StuartsProjects/SX12XX-LoRa #42: Long Range RX improved by a second device close by.

mfvmx, 2022-03-03:
We have a two devices using a Dorji SX1262T modules that are located 1000ft apart, box1 is sending a periodic message once a second. We box2 receives the message from box1 **only** is we have another box3 closer to box2 also broadcasting a message. Once the third box is turned on near box2  the box1-box2 signal is RX'd with no issue at an RSSI ~ -95 and SNR of 12-15. Is there any setting that could be causing this? (Sample Code coming soon)
lora_modem.setupLoRa(FREQUENCY_CENTER_TRACK_STATE, 0, LORA_SF7, LORA_BW_500, LORA_CR_4_5, LDRO_AUTO);

We did a brief test using RadioLib (with the same settings) on box2 and every message was correctly received without the need for box3.

mfvmx, 2022-03-03:
Feather Samd21, sending 35byte messages
Ok Box2 Listening for messages from Box1 with Box3 turned off we get nothing, no responses, no DIO1 pin activity.
We turn on Box3 also sending 35byte messages with a unique identifier and messages from Box3 and Box1 start being received.

mfvmx, 2022-03-03:
So just Box1 transmitting we get :

With Box1 and Box3 both transmitting we get:
14:43:05.103 -> Box3-TestMessage3, RSSI=30, SNR=10
14:43:05.137 -> Box1-TestMessage1, RSSI=97, SNR=15
14:43:06.154 -> Box3-TestMessage3, RSSI=31, SNR=9
14:43:06.189 -> Box1-TestMessage1, RSSI=98, SNR=14
14:43:07.178 -> Box3-TestMessage3, RSSI=28, SNR=11
14:43:07.201 -> Box1-TestMessage1, RSSI=95, SNR=15

vtjballeng, 2022-03-03:
@StuartsProjects , Box 2 never receives messages from Box 1 when only Box 1 & Box 2 are active. 

To receive messages from Box 1 at Box 2, Box 3 (which is closer) must be active. When Box 3 is active, Box 1 messages are then received at Box 2. We record these messages based on the DIO 1 activity, so Box 1 never triggers DIO 1 at Box 2 when only Box 1 & Box 2 are active.

Box 3, which is closer and has a higher signal strength, does trigger DIO 1 at Box 2 and that's when we begin to receive messages from Box 1. 

RadioLib does allow Box 1 to trigger DIO 1 at Box 2. It does not require the presence of Box 3. So it seems there is some fundamental configuration variation between the two libraries. The RadioLib library is problematic on the TX side, with low power TX, which is why we have not been using it.

StuartsProjects, 2022-03-03:
Ignoring for a moment what happens when Box 3 is present, then you appear to be saying that a link between Box 1 and Box 2 works with Radiolib but when the exact same processor and radio modules are used with the SX126X library the link does not work, is this so ?

vtjballeng, 2022-03-03:
@StuartsProjects , yes that is correct.

StuartsProjects, 2022-03-03:
I can run a basic check on the sensitivity differences so I would need to see both sets of code, TX, RX, SX126X and Radiolib.

mfvmx, 2022-03-03:
Without getting into the libraries and easy test is the tree device problem where device3 is increasing the sensitivity of the device2 to all other devices. Create a device1 with a low power output so that if you put it in another room it's above rssi -95, just out of range, and you no longer receive messages from it on device2. Then turn on a device3 near to device2 and you then start receiving messages from device1.

vtjballeng, 2022-03-04:
@StuartsProjects , as @mfvmx mentioned the initial test to validate the issue is fortunately not too burdensome with a few devices. 
 
 I suspect a next step is to peer a little deeper into the registers being set at the SX1262 to see what varies between the two libraries. 
 
 It's also possible this issue is correlated https://forumstg.lora-developers.semtech.com/t/sx1262-reduced-rx-sensitivity-packet-reception-fails/162/8 .

StuartsProjects, 2022-03-04:
I will look at the sensitivity issue at some point.

However you cannot directly work out what has been setup on a SX126x by a register print for instance, as a lot of the config does not seem to appear in registers and documentation of what all the registers really do is non-existent. 

You need to go through the setups manually to work out which commands (and registers) are being written with what values, I can follow the SX126X code easily enough, not at all easy following other libraries.

StuartsProjects, 2022-03-24:
The register write required be datasheet adendum 15.1 is required just before TX is turned on for each packet, the change had been implemented in settx() which the library functions call when sending each packet, this code has been added;

//change required for data sheet addendum 15.1
  uint8_t regvalue = readRegister(REG_TX_MODULATION);
  if (savedModParam2 == LORA_BW_500)
  {
    writeRegister(REG_TX_MODULATION, (regvalue & 0xFB));         //if bandwidth is 500k set bit 2 to 0, see datasheet 15.1
  }
  else
  {
    writeRegister(REG_TX_MODULATION, (regvalue | 0x04));         //if bandwidth is < 500k set bit 2 to 0 see datasheet 15.1.1
  }

With change sensitivity @ 500khz bandwidth improves by around 3dBm.

StuartsProjects, 2023-02-13:
No further action needed

closed 2023-02-13 by StuartsProjects