StuartsProjects/SX12XX-LoRa #43: Soft WDT reset on ESP8285

hitesh1402, 2022-03-08:
Hi. I tried to upload the given code on an ESP8285 instead of a STM32, but I am getting a Soft WDT reset error when not receiving any serial data. While the data is being received, there is no reset, but as soon as the data stops streaming, the error occurs. I've attached my code below. Please have a look, thanks.
[LoRa_Serial_Bridge_Receiver.txt](https://github.com/StuartsProjects/SX12XX-LoRa/files/8204040/LoRa_Serial_Bridge_Receiver.txt)
.

hitesh1402, 2022-03-08:
Sorted out the issue. Sorry for any inconvenience.

closed 2022-03-08 by hitesh1402

StuartsProjects, 2022-03-08:
I dont provide support for the ESP8285.

hitesh1402, 2022-03-10:
Hi,

Thanks for your reply, but the issue was resolved.
Have a great day ahead.

Regards
Hitesh Bhardwaj