StuartsProjects/SX12XX-LoRa #44: [SX1280] Switching from RANGING mode to LORA mode

alphaLeporis, 2022-04-09:
Hi,

I have 2 identical SX1280 modules connected to an ESP32:
1. Sends a regular LoRa message to (2) asking for ranging to start and goes into RANGIN_SLAVE
2. Receives the message and goes into RANGING_MASTER

This seemed easy to do, both codes work seperately. But regular packets don't seem to arrive when switching between them? (TX -> RANGE -> TX (nothing at the receiving end)).

I've tried to do the folowing before going back into transmitting:
```
LT.clearIrqStatus(IRQ_RADIO_ALL); 
LT.resetDevice();
LT.setupLoRa(Frequency, Offset, SpreadingFactor, Bandwidth, CodeRate);
```

Not sure if this is a mistake made by me, or a bug somewhere?

Thanks
Louis

StuartsProjects, 2022-04-09:
Useful to know if the modules stop receiving or transmiting or both, you would need a third SX1280 module for that.

Use the printModemSettings() and printOperatingSettings() to check that setupLoRa() is applying the correct settings. 

I have mixed LoRa and ranging before on balloon trackers, not had a problem, but not tried it on an ESP32. Therfore you would need to see if the same problem occurs on ATmega328 processors, Pro Mini etc.

alphaLeporis, 2022-04-09:
Good to know that it in fact is possible!

When using the printModemSettings() and printOperatingSettings() commands it seems that the correct settings have been used. 

Would you mind sharing some code snippets from your balloon trackers on how you used both? Or isn't it that different from the already provided examples (+/- pasted together)?

Good idea to use a third one, I'm going to try it out! Should I be using the LoRa_Packet_Logger_Receiver?

StuartsProjects, 2022-04-10:
LoRa_Packet_Logger_Receiver is probably the easiest program to use. 

The balloon tracker code, with ranging, was never published.

NobodyNada, 2022-04-10:
I've also been having a lot of trouble when trying to switch a device between ranging slave & lora RX modes, with similar symptoms of packets simply not arriving. The problems seem to be erratic/timing-dependent/difficult to reproduce, and it seems more likely to be an issue with the hardware than anything caused by this library.

However, I discovered that the SX1280 is fully capable of receiving standard LoRa data packets while in ranging-slave mode, so for the time being I hacked together an amalgamation of the `receive` and `receiveRanging` methods:

```c++
int16_t SX128XLT::receiveRangingOrData(uint8_t *rxbuffer, uint8_t size, uint32_t address, uint16_t timeout, int8_t txpower, uint8_t wait)
{
#ifdef SX128XDEBUG
  Serial.println(F("receiveRangingOrData()"));
#endif

  uint8_t index, RXstart, RXend;
  uint16_t regdata;
  uint8_t buffer[2];

  setTxParams(txpower, RADIO_RAMP_02_US);
  setRangingSlaveAddress(address);
  setDioIrqParams(IRQ_RADIO_ALL, (IRQ_RX_DONE + IRQ_RX_TX_TIMEOUT + IRQ_RANGING_SLAVE_RESPONSE_DONE + IRQ_RANGING_SLAVE_REQUEST_DISCARDED + IRQ_HEADER_ERROR), 0, 0);  //set for IRQ on RX done or timeout
  setRx(timeout);

  if (!wait)
  {
    return 0;                          //not wait requested so no packet length to pass
  }

  while (!digitalRead(_RXDonePin));    //Wait for DIO1 to go high

  setMode(MODE_STDBY_RC);              //ensure to stop further packet reception
  regdata = readIrqStatus();

  if ( (regdata & IRQ_HEADER_ERROR) | (regdata & IRQ_CRC_ERROR) | (regdata & IRQ_RX_TX_TIMEOUT ) | (regdata & IRQ_SYNCWORD_ERROR )) //check if any of the preceding IRQs is set
  {
    return -1;                          //packet is errored somewhere so return -1
  }

  if (regdata & IRQ_RANGING_SLAVE_RESPONSE_DONE) {
    return 0;
  }

  readCommand(RADIO_GET_RXBUFFERSTATUS, buffer, 2);
  _RXPacketL = buffer[0];

  if (_RXPacketL > size)               //check passed buffer is big enough for packet
  {
    _RXPacketL = size;                 //truncate packet if not enough space
  }

  RXstart = buffer[1];
  RXend = RXstart + _RXPacketL;
  checkBusy();

#ifdef USE_SPI_TRANSACTION             //to use SPI_TRANSACTION enable define at beginning of CPP file 
  SPI.beginTransaction(SPISettings(LTspeedMaximum, LTdataOrder, LTdataMode));
#endif

  digitalWrite(_NSS, LOW);             //start the burst read
  SPI.transfer(RADIO_READ_BUFFER);
  SPI.transfer(RXstart);
  SPI.transfer(0xFF);

  for (index = RXstart; index < RXend; index++)
  {
    regdata = SPI.transfer(0);
    rxbuffer[index] = regdata;
  }

  digitalWrite(_NSS, HIGH);

#ifdef USE_SPI_TRANSACTION
  SPI.endTransaction();
#endif

  return _RXPacketL;
}
```

I have not tested this extensively but it seems to work for my application. I have one device switching between ranging master & LoRa TX roles and the other device configured as a ranging slave but using this method instead of the standard `receiveRanging` method -- and it seems to work perfectly.

alphaLeporis, 2022-04-13:
Thanks @NobodyNada for your code, I will have a look. But your situation is a a bit different.
I do think there is some setting that should be changed back somewhere. Because, like you mentioned, a Lora packet is not that different than a Ranging packet.

After using the LoRa_Packet_Logger_Receiver code, I can receive something. But it gives "Packet error - IRQ 8052". No clue what that would mean, but I am still looking into it!

StuartsProjects, 2022-04-13:
The IRQs are listed in the SX128XLT_Definitions.h file. 

Or just use the printIrqStatus() function.

StuartsProjects, 2022-04-19:
I have been working on a application that sends a request to a specified station and expects to receice data in return.  

So I modified the sketch to send (with LoRa) a request for a slave to go into ranging mode for a period and the master then sends a number of ranging requests. 

The sketchs quite happily swap between TX and RX LoRa and ranging using standard library functions. 

See ranging examples 223 and 234

alphaLeporis, 2022-04-19:
Thanks, I will be trying them out as soon as possible. 

I briefly looked at the code and it seems that I did the same thing. 

I'll let you know what my results are!

alphaLeporis, 2022-05-12:
Not sure what I did wrong, it is working now!

Thanks!

alphaLeporis, 2022-05-25:
After some more testing I found that the issue could be replicated as following:
Initial setup: 
- 1 Ranging MASTER + TX 
- 1 Ranging SLAVE
- 1 RX node

1. The MASTER will send multiple ranging packets to the SLAVE (works)
2. The MASTER wants to send a regular LoRa packet to the RX node and switches radio's (works)
3. The RX node receives the packet succesfully
4. The MASTER gets back to ranging and transmits these to the SLAVE (does not work - timeout)
After a while the ranging SLAVE function goes into time out and the ranging packets get answered.
It would seem like that the normal LoRa packet gets into the buffer and inhibits the ranging packets to be answered.

After testing the code provided by @NobodyNada (thanks!), I can conclude that this solves the problem. But in my case I do not need to receive LoRa packets.

My question: is this behaviour expected? I am unable to find information about this in the SX1280 datasheet nor the Ebyte e28 manual. 

Thanks!

StuartsProjects, 2022-05-27:
"My question: is this behaviour expected? I am unable to find information about this in the SX1280 datasheet"

Indeed, the datasheet makes no mention of being able to mix Ranging and LoRa packets in the way you seem to be attempting. Switching between LoRa and Ranging does work when the processes described in the datasheet are followed. 

So to find out if the behaviour your seeing is expected, you would need to approach Semtech.