StuartsProjects/SX12XX-LoRa #46: Esp32 don't receive messages

promistrio, 2022-06-24:
Hi! I use esp32 and https://www.ebyte.com/en/product-view-news.html?id=453. Transmitting works well. DIO1 shows that transfer is done. But receive is not working (timeout).
[image]

Transmitter https://github.com/promistrio/beacon
Receiver https://github.com/promistrio/base

StuartsProjects, 2022-06-25:
Which of the SX126X library examples are you using ?

And in particular what is the exact LT.begin(); function you are using ?

And for the record I have never tested that particular LoRa device as I dont have one, or whether my library works under Platform IO.

promistrio, 2022-06-27:
> Which of the SX126X library examples are you using ?
I tested it in arduino studio example SX12XX-LoRa-master/examples/SX126x_examples/ESP32/Basics/103_LoRa_Transmitter_Detailed_Setup_ESP32/Receiver -  not receive.

> And in particular what is the exact LT.begin(); function you are using ?
LT.begin(NSS, NRESET, RFBUSY, DIO1, RX_EN, TX_EN, LORA_DEVICE) 
 and 
LT.begin(NSS, NRESET, RFBUSY, DIO1, SW, LORA_DEVICE)

StuartsProjects, 2022-06-27:
I cannot say what the problem is, the use of RX and TX enable is not supportd on a SX126X device, its mentioned in the documentation;

"Some SX126x modules may have RX or TX enable pins, these are currently not supported by the library"

I have no desire to purchase those particular eByte SX126X modules as they would be illegal to use in my country. 

If someone were to send me a couple, or the money to buy them, I could test them against the libray. 

You could try manually setting RX and TX enable pins before the appropriate functions.

StuartsProjects, 2022-06-27:
In addition its not at all clear from the eByte datasheet exactly how you connect or drive the RX TX enable pins. You would really need to contact them for some sample working code.

promistrio, 2022-06-28:
> I cannot say what the problem is, the use of RX and TX enable is not supportd on a SX126X device, its mentioned in the documentation;
>
> "Some SX126x modules may have RX or TX enable pins, these are [...]

When I used LT.begin with tx_an rx_en I saw a difference on SDR while transmit.

[image]

[image]

promistrio, 2022-06-28:
> "Some SX126x modules may have RX or TX enable pins, these are currently not supported by the library"

I found in library setTX() txEnable functions, which set HIGH on transmit and low on receive

StuartsProjects, 2022-06-29:
I am fully aware that the functions are there, but as mentioned before they are not actually supported as they have never been tested against a LoRa module that needs that type of RX TX switching. 

And as I also mentioned before eByte do not make clear how to do RX TX switching on thier module, you would need to contact them for details.

StuartsProjects, 2022-12-06:
Not clear from ebyte how their module is to be configured.

closed 2022-12-06 by StuartsProjects