StuartsProjects/SX12XX-LoRa #48: Pin settings in settings.h

johncblacker, 2022-07-12:
I'm trying to correlate the settings.h file pin settings with either a esp32-cam board or a arducam esp32s board and am having problems doing so.  Could you provide some help?  What is the "default" board upon which your settings.h file built.  I saw in the readme about the Due...but that doesn't have the camera.  So, I need some help.  Thanks.

StuartsProjects, 2022-07-12:
The readme does mentions that the examples are based on the Pro Mini, mentioned in most of the settings.h files also. 

The readme also mentions that to use the SPI LoRa devices you will need to connect the SPI bus pins, SCK, MOSI and MISO, check the pin diagram for the Arduino you are using for the pin numbers. In this case the pin numbers are fixed and do not need to be specified in the settings.h file. 

The ESP32 can change the pins used for SPI SCK, MOSI and MISO which is why these are defined in the settings.h file for the ESP323CAM examples.

StuartsProjects, 2022-07-12:
Have you looked at the ESP32CAM examples in the \Camera folder ?

johncblacker, 2022-07-12:
ut for example the esp32 cam board bring out IO2, IO4, IO14, IO15, IO13, IO12, IO16 and IO0 + UOR(rx?) and UOT(tx?).
But the settings.h shows connections to 3, 9 and 10 - I can't find these on the esp32-cam board...what am I missing?

StuartsProjects, 2022-07-12:
Which example sketch for the ESP32CAM shows connections for pins 3. 9, 10 ?

StuartsProjects, 2022-07-13:
Program 238_StuartCam_LoRa_Remote_Camera was not tested on an ESP32CAM, as is cleary stated in the notes at the beggining of the program. 

As for programs specifically for an ESP32CAM, they will be the ones where "ESP32CAM" is part of the program name.

johncblacker, 2022-07-13:
OK.  I do have an Arducam to experiment with...thanks for pointing that out...

closed 2022-07-13 by johncblacker