StuartsProjects/SX12XX-LoRa #50: 238_StuartCAM_LoRa_Remote_Camera Compile error

johncblacker, 2022-07-31:
I'm getting an error compiling on my Due.  The error messages are:
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<start messages>>>>>>>>>>>>>>>>
In file included from C:\Users\jobla\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/Arduino.h:31:0,
                 from sketch\238_StuartCAM_LoRa_Remote_Camera.ino.cpp:1:
C:\Users\jobla\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/avr/pgmspace.h:34:0: note: this is the location of the previous definition
 #define PSTR(str) (str)
 ^
In file included from C:\Users\jobla\AppData\Local\Temp\arduino_modified_sketch_665078\238_StuartCAM_LoRa_Remote_Camera.ino:54:0:
C:\Users\jobla\AppData\Local\Temp\arduino_modified_sketch_665078\OV2640.h: In function 'void setupOV2640(uint8_t)':
OV2640.h:39:26: error: 'OV2640_CHIPID_HIGH' was not declared in this scope
     myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
                          ^
OV2640.h:40:26: error: 'OV2640_CHIPID_LOW' was not declared in this scope
     myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
                          ^
exit status 1
'OV2640_CHIPID_HIGH' was not declared in this scope
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<END MESSAGES>>>>>>>>>>>>>>>>>>>>>>>>>>
I'm not sure where the "OV2640_CHIPID_HIGH and LOW" are supposed to be defined...I'm thinking in the Arducam library?  I have picked up the Arducam library off github but maybe not the correct one?  Any ideas?  I'd appreciate your help...
Thanks.

StuartsProjects, 2022-08-01:
When installing the Arducam library, from the link provided in the 238 program code, did you follow the 'How to Use' instructions ?

"How to use

The libraries should be configured before running any examples, or else you will get a compilation error message. 
"

johncblacker, 2022-08-01:
My bad...I don't like modifying libraries so I made a copy in my own script folder but made a mistake...working now, closing.

closed 2022-08-01 by johncblacker