StuartsProjects/SX12XX-LoRa #51: SPI interface error trying to initialize camera

johncblacker, 2022-08-03:
I'm using the Arducam Mini Module 2mp plus connected to Arduino DUE.  I also have an rfm9x from Adafruit and an generic SDcard adaptor connected via SPI.  When the program started, I get the card detected message, then when it tried to setup the OV2640 camera module I get repeated SPI interface errors.  I have put the camera by itself connected to an arduino uno via spi and used the example code from Arducam repository and it works...so I know the cam is OK.  But when I move it over to the DUE I get the SPI interface errors.  I've checked my wiring numerous times and have tried powering the camera with 3.3v and 5v but I still get the error.  Might you have any ideas on how I can trouble shoot this?  From your code it appears you're running the SDcard, camera and LoRa modules on the same SPI bus as am I.  I don't know what to do at this point to troubleshoot and/or fix the problem.

StuartsProjects, 2022-08-04:
Most 'generic' SDcard adapters are not designed to be used with other devices on the SPI bus. 

For a DUE you just need a bare bones SD card holder, the type with no regulator and logic level conversion circuits.

johncblacker, 2022-08-04:
Thanks.  I'll remove the SD card adapter and see where that takes me...Do you have a suggestion on a "bare bones sd card holder?" (Amazon link?).  Thanks again.

johncblacker, 2022-08-04:
I've found the adafruit sd card breakout "ADA254" that uses 3.3v logic.  I'll give it a try.

StuartsProjects, 2022-08-04:
Do a search on eBay\Amazon for 'sd card breakout'

You want a board that is basically just the connector, adapters that have an IC and a regulator, may not work.

johncblacker, 2022-08-05:
OK, thanks.  Switched over to a MKRZERO board with onboard SD and made some progress.  Closing this issue, but please see my other new issue.

johncblacker, 2022-08-05:
Closing awaiting sd card breakout.

closed 2022-08-05 by johncblacker