StuartsProjects/SX12XX-LoRa #52: Errors between program *238*LoRa_Remote* and *241*Receiver

johncblacker, 2022-08-05:
I'm using the MKRZERO board which has an onboard SD card holder/reader/writer.  I'm using example "238" Remote Camera running on the MKRZERO sending (I wish) to example "241" running on Arducam esp32s uno psram.  On the Remote Camera side I get the following messages:
<<<<<<<<<<<<<< 238 remote cam messages >>>>>>>>>>>>>>>>>
Beginning setup
C:\Users\jobla\Documents\Arduino\JB-DUE-ARDUCAM-238_StuartCAM_LoRa_Remote_Camera\JB-DUE-ARDUCAM-238_StuartCAM_LoRa_Remote_Camera.ino
SD Card detected
OV2640 SPI interface OK
OV2640 setup on I2C
LoRa device found

SX1276_PABOOST,SLEEP,Version_12,PacketMode_LoRa,Explicit,CRC_On,AGCauto_On,LNAgain_1
SX1276_PABOOST,915000000hz,SF7,BW500000,CR4:5,LDRO_Off,SyncWord_0x12,IQNormal,Preamble_8

Payload CRC enabled
Start capture 
Done
FIFO length 28172
/1.jpg
/1.jpg Saved to SD card
Send file attempt 1
Start file transfer for /1.jpg
Send open remote file request
NoACK

Send open remote file request
NoACK

Send open remote file request
NoACK

Send open remote file request
NoACK

Send open remote file request
NoACK

********************
/1.jpg
ERROR opening file
Restarting transfer
********************
StartAttempts 1
SDNoAckCount 5
Transmit time 0.000secs
Transmit rate nanbps
<<<<<<<<<<<<<<< end of 238 messages >>>>>>>>>>>>>>>>>>>
And on the example 241 receiver side I'm getting the following messages:
<<<<<<<<<<<<<<<< start of 241 Receiver messages >>>>>>>>>>>>>>>>>>>>
C:\Users\jobla\Documents\Arduino\JB-ARDUCAM-UNO-PSRAM-241_StuartCAM_ESP32CAM_LoRa_Receiver\JB-ARDUCAM-UNO-PSRAM-241_StuartCAM_ESP32CAM_LoRa_Receiver.ino
PSRAM is correctly initialised
PSRAM Size available: 4194252
Allocate array in PSRAM
PSRAM Size available: 4063164
PSRAM array bytes allocated: 131088

LoRa file transfer receiver ready
LoRa device found
Payload CRC enabled
RX Timeout
PacketErrorIRQreg,0x400

PacketErrorIRQreg,0x450

Error receiving picture
Timeout receiving picture

LoRa file transfer receiver ready
LoRa device found
Payload CRC enabled
PacketErrorIRQreg,0x70

Error receiving picture
Timeout receiving picture

LoRa file transfer receiver ready
LoRa device found
Payload CRC enabled
PacketErrorIRQreg,0x70

PacketErrorIRQreg,0x70

Error receiving picture
Timeout receiving picture

LoRa file transfer receiver ready
LoRa device found
Payload CRC enabled
PacketErrorIRQreg,0x450

Error receiving picture
Timeout receiving picture

LoRa file transfer receiver ready
LoRa device found
Payload CRC enabled
PacketErrorIRQreg,0x70
<<<<<<<<<<<<<<<<<< end of 241 output messages >>>>>>>>>>>>>>>>>>>>
Seems like the remote camera side is having trouble opening the file that it had just written after camera capture. So, I popped the card out and there are 10 .jpg files written to it but when I try to open them with "3d paint" the program indicates the file is corrupted.  Eight of the ten files are 1K in size, the other 2 are 3k in size.  I've tried opening several 1k and 3k to no avail.  Any ideas?  When I run the example arduino sd card sketch for the mkrzero it runs fine, but of course it's writting a very small text file then reading it back in and printing the contents.  What do you suggest I do to troubleshoot the issue?

StuartsProjects, 2022-08-05:
I cannot directly assist, the programs and library have not been tested on the particular hardware you are using. It does sound like issues with the RX side SD.  

First test that the LoRa comms is working proplerly and reliably, see the examples in the \Reliable folder. 

Then try some of the programs in the \Data Transfer folder. You can create a small file on the source SD that consists of a known sequence of bytes, such as a text file and see what the received file looks like.

johncblacker, 2022-08-05:
OK, that'll be my next step.  I've gone back to the DUE now since I received my sd breakout board today and installed it.  I'm currently formatting an 8GB sd card and then will give it a go.

johncblacker, 2022-08-05:
Getting very frustrated with this whole setup.  Once I get the sd card working (card detected) then I get spi interface errors trying to get the camera initialized.  Seems like it would be more stable if each device was on a separate SPI interface; but that's a lot of changing in your code.

johncblacker, 2022-08-05:
Closing this and looking for a more stable solution.

closed 2022-08-05 by johncblacker

StuartsProjects, 2022-08-06:
Seperate SPI interfaces for the LoRa and SD card would likely help, however the 238 and 241 examples are very stable on the hardware it was tested on. 

If there are issues with some Arduinos not being reliable when the LoRa and SD card are on the same bus, then the issues will with the hardware used or the SPI and SD libraries that Arduino provide for that particular board.