StuartsProjects/SX12XX-LoRa #53: Message Can't find OV2640 MODULE?

johncblacker, 2022-08-06:
I got to thinking that my breadboards and connections to them might be causing my issues, so I cobbled up a piece of perfboard so that the number of jumper connections were minimized.  After doing so, I'm getting the "Can't find OV2640 Module" message and I've printed the pid/vid which was received and they are both 0.  I did get the "OV2640 SPI interface OK" message, though...ideas?

StuartsProjects, 2022-08-07:
No idea.  

Please appreciate that the software reporting that issue is part of Arducams library, I am not in a position to provide support for that library.

Perhaps try the examples that Arducam provide for their product, with no LoRa device connected.

johncblacker, 2022-08-08:
I think I've spent enough time on this approach. I'm going to go back to my original idea of using a pair of Raspberry Pi's and Python. I'd like to "borrow" a few of your functions and convert them to the micropython/circuit python platform if you wouldn't mind. Of course I'd credit your work as the originator. I was originally looking for a large payload solution when I came upon your repo and then climbed down that rabbit hole aand got frustrated with the whole idea. But, I'm going back to my original plan and adding the necessary functionality for CRC , splitting the image into chunks for transmission, etc. The Pi already has a good cam and usb attached sd and I alreadyhave the Sx1276x/rfm9x driver code working. I appreciateyour patience with my previous efforts. Please let me know of your decision about using yout code as a base for reliable, large payload transmission.

⁣Get BlueMail for Android ​

***@***.***>

StuartsProjects, 2022-08-08:
I don’t think the problems you are seeing are related to the Arducam library or my LoRa library. 

My own LoRa library follows the appropriate methods for using the SPI library and the Arducam, SD and LoRa data transfer code is stable on the same SPI bus on a Arduino DUE, ATmega328P and ATmega1284P cores. 

However there are circumstances where I have found the SPI and SD libraries of a core not working well together. This happens on an ESP32 for instance. My file transfer code does work on an ESP32 when the LoRa and SD card share the SPI bus. However its not stable and the SD card stops responding after a while. But if the SD card is used on the ESP32 in MMC mode, which does not use the SPI bus pins, its stable. Ergo there seem to be some issue with  the SPI or SD library on the ESP32.

The ESP32CAM does use the SD card in MMC code and the LoRa image file transfer stuff works OK on this simple board. But the camera is not so good. 

Then there is the issue of using the wrong type of SD card adapter. The common cheap ones, that have a regulator and level translator circuit, don’t work with other components on the SPI bus. There is a flaw in the design of these common boards that means they do not release the SPI bus correctly. 

Raise a contact request on the StuartsProjects website and I can give you some pointers about doing such image\file transfers on the Pi.

johncblacker, 2022-08-09:
I don't know the address of your website nor how to "raise a contact request" regarding image/file transfers on the Pi.

StuartsProjects, 2022-08-11:
Is on the same Github location as this library;

https://stuartsprojects.github.io/

StuartsProjects, 2022-10-05:
Not an issue wit the library code or examples

closed 2022-10-05 by StuartsProjects