StuartsProjects/SX12XX-LoRa #54: Compile errors with 238_StuartCAM_LoRa_Remote_Camera using Arducam_esp32s_uno_psram board

johncblacker, 2022-08-10:
I'm wanting to get the subject sketch to compile and run on the Arducam esp32s uno psram board, primarily because it already has an SD card on the board.  However, I'm getting compile errors that appear related to esp32 xtensa tools functions.  Here's my compile output:
<<<<<<<<<<<<<<<<<< compile output log >>>>>>>>>>>>>>>>>>>>>>>>
In file included from c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:39:0,
                 from c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\functional:55,
                 from c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\memory:79,
                 from C:\Users\jobla\AppData\Local\Arduino15\packages\ArduCAM_ESP32S\hardware\esp32\1.0.0\libraries\FS\src/FS.h:24,
                 from C:\Users\jobla\AppData\Local\Arduino15\packages\ArduCAM_ESP32S\hardware\esp32\1.0.0\libraries\SD\src/SD.h:17,
                 from C:\Users\jobla\Documents\Arduino\libraries\SX12XX-LoRa-master\src/DTSDlibrary.h:15,
                 from C:\Users\jobla\Documents\Arduino\JB-ARDUCAM-UNO-ESP32S-PSRAM-238_StuartCAM_LoRa_Remote_Camera\JB-ARDUCAM-UNO-ESP32S-PSRAM-238_StuartCAM_LoRa_Remote_Camera.ino:48:
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\array:115:26: error: macro "swap" requires 3 arguments, but only 1 given
       swap(array& __other)
                          ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\array:116:72: error: macro "swap" requires 3 arguments, but only 2 given
       noexcept(noexcept(swap(std::declval<_Tp&>(), std::declval<_Tp&>())))
                                                                        ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\array:274:56: error: macro "swap" passed 4 arguments, but takes just 3
     swap(array<_Tp, _Nm>& __one, array<_Tp, _Nm>& __two)
                                                        ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\array:275:39: error: macro "swap" requires 3 arguments, but only 1 given
     noexcept(noexcept(__one.swap(__two)))
                                       ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\array:276:23: error: macro "swap" requires 3 arguments, but only 1 given
     { __one.swap(__two); }
                       ^
In file included from c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\functional:55:0,
                 from c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\memory:79,
                 from C:\Users\jobla\AppData\Local\Arduino15\packages\ArduCAM_ESP32S\hardware\esp32\1.0.0\libraries\FS\src/FS.h:24,
                 from C:\Users\jobla\AppData\Local\Arduino15\packages\ArduCAM_ESP32S\hardware\esp32\1.0.0\libraries\SD\src/SD.h:17,
                 from C:\Users\jobla\Documents\Arduino\libraries\SX12XX-LoRa-master\src/DTSDlibrary.h:15,
                 from C:\Users\jobla\Documents\Arduino\JB-ARDUCAM-UNO-ESP32S-PSRAM-238_StuartCAM_LoRa_Remote_Camera\JB-ARDUCAM-UNO-ESP32S-PSRAM-238_StuartCAM_LoRa_Remote_Camera.ino:48:
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:328:31: error: macro "swap" requires 3 arguments, but only 2 given
         std::declval<_Head&>()))
                               ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:332:36: error: macro "swap" requires 3 arguments, but only 2 given
  swap(_M_head(*this), _M_head(__in));
                                    ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:454:76: error: macro "swap" requires 3 arguments, but only 2 given
       noexcept(noexcept(swap(std::declval<_Head&>(), std::declval<_Head&>())))
                                                                            ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:457:36: error: macro "swap" requires 3 arguments, but only 2 given
  swap(_M_head(*this), _M_head(__in));
                                    ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:580:23: error: macro "swap" requires 3 arguments, but only 1 given
       swap(tuple& __in)
                       ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:590:23: error: macro "swap" requires 3 arguments, but only 1 given
       void swap(tuple&) noexcept { /* no-op */ }
                       ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:740:23: error: macro "swap" requires 3 arguments, but only 1 given
       swap(tuple& __in)
                       ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:1132:60: error: macro "swap" requires 3 arguments, but only 2 given
     swap(tuple<_Elements...>& __x, tuple<_Elements...>& __y)
                                                            ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:1133:35: error: macro "swap" requires 3 arguments, but only 1 given
     noexcept(noexcept(__x.swap(__y)))
                                   ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\tuple:1134:19: error: macro "swap" requires 3 arguments, but only 1 given
     { __x.swap(__y); }
                   ^
In file included from c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\memory:79:0,
                 from C:\Users\jobla\AppData\Local\Arduino15\packages\ArduCAM_ESP32S\hardware\esp32\1.0.0\libraries\FS\src/FS.h:24,
                 from C:\Users\jobla\AppData\Local\Arduino15\packages\ArduCAM_ESP32S\hardware\esp32\1.0.0\libraries\SD\src/SD.h:17,
                 from C:\Users\jobla\Documents\Arduino\libraries\SX12XX-LoRa-master\src/DTSDlibrary.h:15,
                 from C:\Users\jobla\Documents\Arduino\JB-ARDUCAM-UNO-ESP32S-PSRAM-238_StuartCAM_LoRa_Remote_Camera\JB-ARDUCAM-UNO-ESP32S-PSRAM-238_StuartCAM_LoRa_Remote_Camera.ino:48:
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\functional:2035:16: error: macro "swap" requires 3 arguments, but only 1 given
  __x.swap(*this);
                ^
c:\users\jobla\appdata\local\arduino15\packages\arducam_esp32s\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\functional:2075:26: error: macro "swap" requires 3 arguments, but only 1 given
  function(__x).swap(*this);
[... 132 lines cut ...]
                 from C:\Users\jobla\AppData\Local\Arduino15\packages\ArduCAM_ESP32S\hardware\esp32\1.0.0\libraries\FS\src/FS.h:24,
[... cut ...] ludes FS.h which is where the errors emanate from.  So, I can't figure out why the same error(s) wouldn't show up with your code, say on a DUE which includes SD.h that also includes FS.h.  But, of course, the SD.h library on the DUE or plain Arduino platform would be getting different versions of SD.h and FS.h.  But, when compiling the original ARDUCAM example script, which is using the ESP32 platform, I would have expected the same errors to occur, but they don't.  Any ideas about this problem.

johncblacker, 2022-08-10:
I have a quick question for you...do you have the ARDUCAM 32S board manager version 1.0.0 installed or did you find a later version?

StuartsProjects, 2022-08-11:
Looks like the Arducam library does not support that ESP32 board. 

You would need to contact Arducam to sort that out.

StuartsProjects, 2022-08-18:
Not an issue with the SX12xx library

closed 2022-08-18 by StuartsProjects