StuartsProjects/SX12XX-LoRa #55: ESP32 receiver program 241Picture write errors

johncblacker, 2022-08-17:
I'm making progress...at least my transmitter (program 240 with the esp32 (arducam esp32s uno psram) is writing the pictures to the SD; but the receiver is not, I get the following messages on each file write:
<<<<<<<<<<<<<<<<< start of log message from write >>>>>>>>>>>>>>>>
/pic74.jpg Save picture to SD card
E (1852371) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (1852371) diskio_sdmmc: sdmmc_write_blocks failed (265)
/pic74.jpg Saved to SD
E (1852374) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (1852381) diskio_sdmmc: sdmmc_write_blocks failed (265)
<<<<<<<<<<<<<<<< end of log messages >>>>>>>>>>>>>>>>>>>>>>>>>>>
Receiver is also running on Arducam ESP32S_UNO_PSRAM with built-in SD card.  I'm using 8gb cards on both machines.  Do you have any ideas?  Thanks.

johncblacker, 2022-08-17:
I got both recvr and xmtr to write to the sd cards...switched the functions around between the two machines and had to remove and insert the sd card several times before the program was able to read/write to them.  Very flaky and unpredictable, at best.  I'm thinking this kind of unstable operation isn't going to cut if for me and I'm going to move over to the rPi using Adafruit LoRa bonnets with code written in Python...I'll have to add my own code to split the images into 251 byte segments and then put them back together but it's worth the time to do that given how much time I've spend already on trying this with the Due and Arducam platforms using C/C++.  However, I do appreciate your patience and the work that you put into your github repository.  I'm glad it works for you, and probably countless others, just not for me.

johncblacker, 2022-08-17:
Great work but should look at porting over to the rPi for more stability and better camera support.

StuartsProjects, 2022-08-18:
Program 241 was explicitly written for the ESP32CAM, it would not directly port to other hardaware. 

The programs published are relaible on the hardware indicated, There are examples of the image\file transfer for the ESP32CAM and NodeMCU, both are low cost boards, particularly the ESP32CAM that even has a built in SD card reader. 

You might

StuartsProjects, 2022-10-05:
Not an issue with published examples.

closed 2022-10-05 by StuartsProjects