StuartsProjects/SX12XX-LoRa #56: Inconsistent power usage

BioScout-Tom, 2022-10-29:
Hello,

Firstly, thanks for the fantastic library Stuart. It has truly made LoRa implementation (particularly as one of the few references for 2.4GHz) more accessible to the community!

I am implementing a LoRa sensor node, and experiencing a bizarre behaviour with regards to transmit current consumption. The behaviour I am seeing is:

- A Large 300mA current spike for the first transmission, and every second transmission thereafter. (Power Trace in 
[image]

I am using a version of the ESP32 example modified for my pin configuration

Arduino Sketch:

`/*******************************************************************************************************
  Programs for Arduino - Copyright of the author Stuart Robinson - 04/04/20

  This program is supplied as is, it is up to the user of the program to decide if the program is
  suitable for the intended purpose and free from errors. 
*******************************************************************************************************/

/*******************************************************************************************************
  Program Operation - This is a program that demonstrates the detailed setup of a LoRa test transmitter. 
  A packet containing ASCII text is sent according to the frequency and LoRa settings specified in the
  'Settings.h' file. The pins to access the lora device need to be defined in the 'Settings.h' file also.

  The details of the packet sent and any errors are shown on the Arduino IDE Serial Monitor, together with
  the transmit power used, the packet length and the CRC of the packet. The matching receive program,
  '104_LoRa_Receiver' can be used to check the packets are being sent correctly, the frequency and LoRa
  settings (in Settings.h) must be the same for the transmitter and receiver programs. Sample Serial
  Monitor output;

  10dBm Packet> Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,64mS  PacketsSent,2

  Serial monitor baud rate is set at 9600
*******************************************************************************************************/

#define Program_Version "V1.1"

#include <SPI.h>                                               //the lora device is SPI based so load the SPI library                                         
#include <SX128XLT.h>                                          //include the appropriate library  
#include "Settings.h"                                          //include the setiings file, frequencies, LoRa settings etc   

SX128XLT LT;                                                   //create a library class instance called LT

uint8_t TXPacketL;
uint32_t TXPacketCount, startmS, endmS;

uint8_t buff[] = "Hello World 1234567890";

void loop()
{
  //LT.wake();
  
  TXPacketL = sizeof(buff);                                    //set TXPacketL to length of array
  buff[TXPacketL - 1] = '*';                                   //replace null character at buffer end so its visible on reciver
  
  LT.printASCIIPacket(buff, TXPacketL);                        //print the buffer (the sent packet) as ASCII

  startmS =  millis();                                         //start transmit timer
  if (LT.transmit(buff, TXPacketL, 10000, TXpower, WAIT_TX))   //will return packet length sent if OK, otherwise 0 if transmit error
  {
    endmS = millis();                                          //packet sent, note end time
    TXPacketCount++;
    packet_is_OK();
  }
  else
  {
    packet_is_Error();                                 //transmit packet returned 0, there was an error
  }

[... 122 lines cut ...]
//#define RFBUSY 25                               //busy line

#define LORA_DEVICE DEVICE_SX1280               //we need to define the device we are using
#define NSS 34                                 //select pin on LoRa device
#define NRESET 21                                //reset pin on LoRa device
#define RFBUSY 26                                //busy pin on LoRa device 
#define DIO1 33                                  //DIO1 pin on LoRa device, used for sensing RX and TX done
#define DIO2 -1                                 //DIO2 pin on LoRa device, normally not used so set to -1 
#define DIO3 -1                                 //DIO3 pin on LoRa device, normally not used so set to -1
#define RX_EN -1                                //pin for RX enable, used on some SX128X devices, set to -1 if not used
#define TX_EN -1                                //pin for TX enable, used on some SX128X devices, set to -1 if not used 
#define BUZZER -1                               //pin for buzzer, set to -1 if not used 

//*******  Setup LoRa Parameters Here ! ***************

//LoRa Modem Parameters
const uint32_t Frequency = 2445000000;          //frequency of transmissions
const int32_t Offset = 0;                       //offset frequency for calibration purposes  
const uint8_t Bandwidth = LORA_BW_0400;         //LoRa bandwidth
const uint8_t SpreadingFactor = LORA_SF12;       //LoRa spreading factor
const uint8_t CodeRate = LORA_CR_4_5;           //LoRa coding rate

const int8_t TXpower = 10;                      //LoRa transmit power in dBm

const uint16_t packet_delay = 3000;             //mS delay between packets`

StuartsProjects, 2022-10-29:
There have been no issues reported, caused by the library, when power consumption in transmit is so high.

If you lengthen the packet sent to a couple of seconds, you will be able to measure the current consumption with a multimeter. 

You included two copies of the program you are using in the ZIP file, and the newer one (29\10\22) does not appear to be for an ESP32 as some of the pins used as outputs are input only.

BioScout-Tom, 2022-10-29:
Sorry about the files, the most recent one is the one I am working with and it is for an esp32s2-mini-1 (no PSRAM). I understand that isn't officially supported but I am keen to work it out. I will dig further into all of my signals with a scope and see if there is any funny business going on. 

To your knowledge is there anything in the library that may flip-flop in this way? 

I have just repeated testing on an MKR NB 1500 and a LAMBDA80 Sx1280 module, and found that it doesn't display the same issue:

[image]

StuartsProjects, 2022-10-29:
Thats not behaviour I have seen before. I cannot see how you could configure the SX1280 to suddenly use 5 or 6 times the amount of current it would normally use at the maximum confugurable power output of 12.5dBm.

Which SX1280 module are you using on the ESP32 S2 ?

What are the RSSI readings for a receiver picking up the low current and high current packets ? 

"I have just repeated testing on an MKR NB 1500 and a LAMBDA80 Sx1280 module, and found that it doesn't display the same issue:"

Which library were you using ?

BioScout-Tom, 2022-10-29:
For the MKR NB 1500 testing, I was using the non-esp32 equivalent example program from your library, with only pin allocation changed. 
 
[MKRNB1500_103_LoRa_Transmitter_Detailed_Setup.zip](https://github.com/StuartsProjects/SX12XX-LoRa/files/9893662/MKRNB1500_103_LoRa_Transmitter_Detailed_Setup.zip)

Qualitatively, I think the RSSI readings are higher for the larger current spikes, and the lower current messages are prone to reporting an error on Rx side, (and unsuccessful ACK on Tx side when using reliable comms functions).

The ESP32-S2 is on a custom PCB with my own implementation of the SX1280 reference design, I used the same schematic as that used in the ExpressLRS project:
[image]

As you say, that much current has to be going somewhere. I modified my code to send 500mS packets as quickly as possible, and indeed the SX1280 chip itself starts heating, giving a terrifying square-wave from 100-300mA. At this point I for some reason I ripped out the filter from the RF output of the SX1280 with snips, and connected the antenna directly to the output of the chip. This resolved the power draw issue.

Here is the component seemingly at fault: [tdk dea102450](https://product.tdk.com/en/system/files?file=dam/doc/product/rf/rf/filter/catalog/rf_bpf_dea102450bt-1278a2_en.pdf)

Happy its resolved, but very confused. I will test across some further hardware to see if that component is simply not correct, as I have some Semtech-endorsed [2450FM07D0034T](https://www.digikey.com.au/en/products/detail/johanson-technology-inc/2450FM07D0034T/9477815), that I can replace it with. If the behaviour persists with that component, then there might actually be a really interesting bug going on. Otherwise, happy to call it a hardware error.

Thanks for your time Stuart. As some small return insight, happy to let you know that I've now used the library across ESP8266, ESP8285, ESP32-E, ESP32S2, NRF52840, as well as been successfully able to port the library to the Particle IDE and use it with a Particle Boron.

StuartsProjects, 2022-10-30:
No issue found with library.

closed 2022-10-30 by StuartsProjects