StuartsProjects/SX12XX-LoRa #58: 241_StuartCAM_ESP32CAM_LoRa_Receiver issue

asadi2021, 2022-12-05:
Hi, 
I am working with 241_StuartCAM_ESP32CAM_LoRa_Receiver example and it is work without any problem, but I can not read images in SD Card , I use this codes :

  SPI.end();
  digitalWrite(NSS, HIGH);
  digitalWrite(NRESET, HIGH);

But I get the following error:
SD CARD OK
E (56431) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (56431) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (56432) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (57437) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (57437) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (57438) diskio_sdmmc: sdmmc_read_blocks failed (263)
Opening file failed ....
Please help me
thank you.

StuartsProjects, 2022-12-08:
Did you remove the transistor controlling the white LED ?

Apart from that it sounds like a problem with the SD card, normally best to remove it whilst programming the ESP32CAM and then powering the ESP32CAM off put the SD card back in and power on again, thats all I can suggest, the program works fine here.

StuartsProjects, 2023-02-13:
No issues with library.

closed 2023-02-13 by StuartsProjects