StuartsProjects/SX12XX-LoRa #59: Send and recieve data in one board ESP32CAM and SX1278

asadi2021, 2022-12-07:
Hi ,
Is there a difference between the settings of the receiver and the transmitter? How can I receive data from the receiver before sending the image in sender board?
Is it correct to use the following sending and receiving codes?
For example, in examples 240 and 241 : 

In Receiver Board :
if (setupLoRaDevice())
  {
sentOK = ARsendArray(Data_buf, Data_len, Code1, tempsize + 1); 
if(sentOK ) {
arraylength = ARreceiveArray(PSRAMptr, sizeof(ARDTarraysize), ReceiveTimeoutmS);
.
.
.
/* some code */
.
.
}
}

In Trasmitter Board : 
if (setupLoRaDevice())
{
arraylength = ARreceiveArray(PSRAMptr, sizeof(ARDTarraysize), ReceiveTimeoutmS);
if (arraylength ){
sentOK = ARsendArray(fb->buf, fb->len, filenamearray, tempsize + 1);
.
.
.
/* some code */
.
.
}
}

StuartsProjects, 2022-12-08:
For the camera programs the terms 'transmitter' and 'receiver' refer to the transmission and reception of the picture. 

The 'transmitter' program sends a packet and then switches to receive mode to receive the acknowledgment. The 'receiver' program when it receives a packet switches to transmit mode to send the acknowledgment. 

So both programs are aleady a working example of using a single ESP32CAM as a LoRa transmitter and receiver.

StuartsProjects, 2023-02-13:
No issues with library.

closed 2023-02-13 by StuartsProjects