StuartsProjects/SX12XX-LoRa #63: Dual E28-2g4m27s with the same board

Xarin94, 2023-02-13:
Hi, I'm trying to use two modules on the same spi bus, one module is sending a flrc packet, the other one should be listening.
When testing the other module receives only sporadic noise, no packets.
Same freq and params, there is a compatibility issue with two module operation and this library?
After many tries i modified a little an example to have low chance of mistakes.
[Example.pdf](https://github.com/StuartsProjects/SX12XX-LoRa/files/10720979/Example.pdf)

StuartsProjects, 2023-02-13:
Having a transmitter next to a receiver is likely to cause problems in any case. 

The library was not designed to allow two instances on the same Arduino.

Xarin94, 2023-02-13:
The transmitter has a attenuator filter on the antenna, i'm using the ipex ones, the teensy 4.0 is capable of handling the workload, my goal is to have a transitter ad reciver with full duplex capability.
in this care there is a way to separate the shared parameters?

StuartsProjects, 2023-02-13:
I have no idea sorry, never tried it.

Xarin94, 2023-02-15:
trying with radiolib

closed 2023-02-15 by Xarin94