StuartsProjects/SX12XX-LoRa #64: LoRa + SD (SPI issues)

TigoTas, 2023-02-17:
Hi, 

I am using the Heltec WiFi LoRa 32 (V2), plus a SD breakout as this one:

https://www.makerhero.com/produto/modulo-cartao-micro-sd/?utm_source=google&utm_medium=organic&utm_campaign=shopping&utm_content=surfaces_across_google&gclid=Cj0KCQiA6LyfBhC3ARIsAG4gkF8YpkuXg6qaXBVLntSmTtUdMAicUc2-Ub3veN0M6yr7a8Ot7GryhMEaAmhzEALw_wcB

The Heltec board has the default SPI pins as:

MISO 19
MOSI 27
CLK 5
CS 18

In te code from the "233" example, I made sure to set the pins in the settings file:

#define NSS    18 //select pin on LoRa device
#define NRESET 14 //reset pin on LoRa device
#define DIO0   26 //DIO0 pin on LoRa device, used for sensing RX and TX done 
#define LED1   25 //LED used to indicate transmission
#define SDCS   13 //SD card Chip Select pin 

I can't get the SD to work with the LoRa. 

I tried several "fixes", but no luck so far.

I noticed the following:

- If I run the code with everything connected, it won't start LoRa and I get the "LoRa device error";
- If I remove the MISO pin from the SD interface, LoRa starts succesfully, but then I get the "SD card failed...or not present" message;

Any help is appreciated.

TigoTas, 2023-02-17:
I also tried the 2 SPIs approach. But if I, for instance, declare a vspi class with the default pins for the LoRa and a hspi class for the SD, I can't get none of them to work.

```
#define NSS    18 //select pin on LoRa device
#define NRESET 14 //reset pin on LoRa device
#define DIO0   26 //DIO0 pin on LoRa device, used for sensing RX and TX done
#define LED1   25 //LED used to indicate transmission
#define SDCS   13 //SD card Chip Select pin

#define VSPI_CLK  5
#define VSPI_MISO 19
#define VSPI_MOSI 27
//#define VSPI_CS   18 //Use NSS

#define HSPI_CLK  14
#define HSPI_MISO 12
#define HSPI_MOSI 13
//#define HSPI_CS   13 //Use SDCS

SPIClass vspi = SPIClass(VSPI);
SPIClass hspi = SPIClass(HSPI);

void setup()
{
  pinMode(LED1, OUTPUT);                          //setup pin as output for indicator LED
  pinMode(NSS, OUTPUT); 
  pinMode(SDCS, OUTPUT); 

  led_Flash(2, 125);                              //two quick LED flashes to indicate program start
  
  SDsetLED(LED1);                                 //setup LED pin for data transfer indicator

  digitalWrite(NSS, HIGH);
  digitalWrite(SDCS, HIGH);

  #ifdef ENABLEMONITOR
    Monitorport.begin(115200);
    Monitorport.println();
    Monitorport.println(F(__FILE__));
    Monitorport.flush();
  #endif

  vspi.begin(VSPI_CLK, VSPI_MISO, VSPI_MOSI, NSS);
  hspi.begin(HSPI_CLK, HSPI_MISO, HSPI_MOSI, SDCS);  
  //SPI.begin();

  digitalWrite(NSS, LOW);

  if (LoRa.begin(NSS, NRESET, DIO0, LORA_DEVICE))
    {
      led_Flash(2, 125);
    }
  else
    {
      #ifdef ENABLEMONITOR
        Monitorport.println(F("LoRa device error"));
      #endif
      
      while (1)
        {
          led_Flash(50, 50); //long fast speed flash indicates device error
        }
    }
```

StuartsProjects, 2023-02-18:
This is where issues with the Library code or examples are raised and there is no suggestion that there is a fault there.

Your using an SD card adapter that is desiged for 5V logic Arduinos, the Heltec board is a 3.3V logic Arduino. In addition that type of SD card adapter is well known not to work with other SPI devices on the same bus. 

You need to use a appropriate SD card adapter.

TigoTas, 2023-02-22:
You are entirely right. Got it working with a different breakout board for the SD card.

But some of my original issues persist.

In your code, you use the default SPI pins. For some reason, if I create a new instance of SPI with different pins, either for LoRa, or the SD, it won't work.

Any helpful tips for this one?

StuartsProjects, 2023-02-22:
The code published, for the example you quoted, was for an Arduino DUE, as mentioned in the notes. The DUE does not allow re-direction of the SPI bus pins, so you have no choice on which pins to use. 

I have not tried to run my library code with a two SPI bus Arduino.

TigoTas, 2023-02-22:
Understood.

Thank you very much for the prompt answers.

closed 2023-02-22 by TigoTas