StuartsProjects/SX12XX-LoRa #65: Add support for setting output stream

arduino12, 2023-05-07:
**Hi,**

**This library is awesome! many thanks!**

I use it with my ESP32-S3 MCU- it have multiple UART peripherals (many `Serial` instances in the Arduino environment)..
I specifically use the Virtual Serial Port (VCP) with `USBSerial.begin(0); // baudrate have no meaning on VCP...`
and `USBSerial.println("LoRa!");` for example.
(Also [Arduino Micro](https://store.arduino.cc/products/arduino-micro) and [Arduino Mega2560](https://store.arduino.cc/products/arduino-mega-2560-rev3) have many `Serial` instances).

My problem is with the hard-coded `Serial.print` in this library - I cannot set it to print to other `Serial` instances..

1. **I think the best solution is replacing all `Serial.print` with `_stream.print`, and add:**
`LT.setLogStream(Stream &stream);` that allows setting the output stream like I did [here](https://github.com/arduino12/rdm6300/blob/master/src/rdm6300.h#L45).

2. Another option is using a macro in the .h file like (replacing all `Serial.print` with `SERIAL_DEV.print`):
```
#ifndef SERIAL_DEV
	#define SERIAL_DEV	Serial
#endif
...
SERIAL_DEV.println(...);
```
So users can choose to define their own stream like:
```
#define SERIAL_DEV	USBSerial
#include <SX128XLT.h>
```
But this one isn't very elegant..

3. Another option is like the first one - but every function that print something will get the stream as an argument-
So no need to add another `LT.setLogStream` function - but I think its better to just save an internal stream pointer..

Many libraries allows setting the I2C or SPI or UART streams, I also see an open [PR 33](https://github.com/StuartsProjects/SX12XX-LoRa/pull/33) for SPI.

StuartsProjects, 2023-05-07:
I currently have no plans at present to make such a change. 

Whilst the change itself might not be seen as time consuming, it would have to be made for the SX126x, SX127X and SX128X libraries. Checking that the change had been implemented correctly and that examples performed correctly would take a considerable amount of testing.

arduino12, 2023-05-07:
I have SX1280 so I can test its examples, maybe other who sees this issue and want that feature can offer testing :)

But if you go with my #2 implementing option (using a macro) - it is 100% backwards compatible down to the final compiled .hex file-
So testing can be done by comparing the compiled binaries before and after the change and make sure they are the same using a script (that script can be re-used to make sure all examples compiles after every small commit as well).
(I once wrote similar script using [Arduino cli](https://arduino.github.io/arduino-cli/0.32/)).

Anyway thanks for your quick reply!