StuartsProjects/SX12XX-LoRa #68: Any reason this isn't in the arduino/platformio package ecosystem?

factoidforrest, 2023-08-06:
This library seems fantastic. Any reason it's not packaged up for easy use in those systems?

Using it as a zip is a little inconvenient and platformio is pushing people away from that.

StuartsProjects, 2023-08-07:
I have not checked that its compatible with the Arduino Library manager. 

I have been considering changing the library so it supports multiple SPI and Serial ports, but all the checking of examples across the 3 platforms is a significant amount of work, and not a priority task at the moment. 

If the library works in platform IO then fine, but I wont be checking that it does.

factoidforrest, 2023-08-09:
It does (I'm using it), and it would be trivially easy to add it to the registry. Simply running `pio pkg publish` will do the job, I think!

I wonder how difficult it would be to get that into a github action so it happens automatically. Let me know if you'd consider a contribution for that.

I would think increasing the number of people benefitting from this excellent library would be one of your chief goals!

StuartsProjects, 2023-10-03:
Currently no plans to check library within platformio

closed 2023-10-03 by StuartsProjects